jnavila/plotkicadsch #2: Compilation on macOS

Gasman2014, 2017-07-03:
I have managed to compile this on macOS although I have had a few problems. :(

 The initial compile failed with a `git.unix` dependency issue although I have `git-unix` installed and updated. I changed  `git.unix` to `git-unix` in the _oasis file and I could then compile and install OK. However, I am now getting `Fatal error: exception Plotgitsch.InternalGitError("path not found: /")` when I run `plotgitsch` from within a valid git checkout containing a Kicad project .pro & schematic .sch with a committed revision history.

Any advice?

jnavila, 2017-07-04:
I developped this on a debian testing system and it worked. When ported to Ubuntu, it failed with the same error. I'll try fixing it, stay tuned.

Gasman2014, 2017-07-04:
Thank you - I was also trying an alternative (eeshow https://neo900.org/stuff/eeshow/ )but that just spirals into an annoying dependency hell related to problems building for x-windows on macOS. Would prefer not to have to run _everything_ in a virtual environment!

Gasman2014, 2017-09-12:
Did you have any further thoughts on this? I have been writing a visual diff for Kicad pcbnew with a web output https://github.com/Gasman2014/KiCad-Diff  but would quite like to get something similar working for schematics. Your notes on smudge/clean were very helpful to me.

jnavila, 2017-09-13:
The export to svg of already existing schematic files is still working (plotkicadsch), but the git library for extracting the files at a given reference has undergone a lot of changes and I hadn't much time to try to fix it.

Gasman2014, 2017-09-13:
OK, thanks for the hint - I will have a look and see if I can bend it to my needs. Actually, I don’t want to use git anyway - I would quite like to get the versioning aspect to pull out Fossil scm versions (https://www.fossil-scm.org/index.html/doc/trunk/www/index.wiki) as it has more integrated functionality for my plans.

BW

John

jnavila, 2017-09-15:
JFTR, I fixed the ocaml-git interface. I had a look at your graphical comparison script and it seems your image magick scripts are much more advanced than mine. I will maybe borrow you some ideas. Thanks.

Gasman2014, 2017-09-17:
Thank you - I will have a go again tomorrow with the ocaml-git interface - just been trying to sort out some python related issues in a new Kicad install :(….

Not sure about ‘advanced’ but hope they work for you - you are very welcome to borrow (as I did from you!)

One of the issues I have struggled with is the seeming random area that pcbnew plots.  I am hitting a problem that if you add something OUTSIDE the current board outline, the plot area is adjusted - so it throws off any visual diff. I will have another play with it in a week or two when I get time.

BW

John

Gasman2014, 2017-10-13:
Managed to get this to compile and run now on macOS - Thank you!.  The main issues now are that the font is fixed on TimesNew Roman and I am having problems with hierarchical sheets but I will keep having a play with it (had a bit of a major laptop disaster last week involving  a new keyboard replacement in my MacBook Pro - LOTS of small screws).
BW

John

jnavila, 2017-10-13:
Hi @Gasman2014 

I have a few questions concerning your last comment:

 * I didn't specify the font to be Times New Roman. The generated svg file does not contain any reference to a font in fact. Only font sizes are set up according to what's available in the sch file. Where are you seeing this? Can you attach a screen copy?
 * Right now the program is able to manage hierarchical sheets, but it is not able to differentiate which one of multiple instances of a sheet was changed: it's only a file based difference, and the default references for components are used. Is it the problem you are referring to?

Jean-Noël

Gasman2014, 2017-10-14:
The font thing is not a major problem but here are a couple of pairs of screenshots;

[image]
[image]
[image]
[image]

There are some minor positioning errors - the input label, for instance, is on the other side.
Ignore the hierarchical sheet comment - I was failing to load in the necessary library.

Gasman2014, 2017-10-16:
> Text block alignments are not correctly managed when there are rotations or mirrors.

Hi Jean-Noël,

I am pretty sure that the text alignment is OK for the rotated components - I was meaning the 'Input' is the wrong side of the line (see circled item).Providing it is consistent, it isn't a problem, I guess. 

Any idea on the font issue? How are fonts specified? I think Kicad uses its own internal font and renders it in a nonstandard way.

BW

John

jnavila, 2017-10-27:
Just a heads up to say that the texts should now be correctly handled.

Relying on the default font is not an option. Which font would you prefer?

Gasman2014, 2017-10-28:
Hiya

That sounds good!

The kicad default font is a monospaced sans serif. It is also a single line font - like an engraving font. I guess that something similar as possible  would be best.
I tried out a few samples on various font matching sites and think that, of the ttf fonts, something like Ubuntu mono is a pretty good match (and has a suitable licence).

Alternatively, there are a number of Hershey engraving fonts which are ’single stroke’ (http://climserv.ipsl.polytechnique.fr/documentation/idl_help/Hershey_Vector_Font_Samples.html <http://climserv.ipsl.polytechnique.fr/documentation/idl_help/Hershey_Vector_Font_Samples.html>)
These are available in a off .cxf format - I have used these on my cnc machine- they are installed with inkscape, I think. one of these might also be appropriate? I don’t think there are any licensing issues with these either.

BW

John

jnavila, 2017-11-09:
For the fonts, I just chose a mono font. The one used in Kicad cannot be found as a system one. Did you notice any difference?

Gasman2014, 2017-11-10:
Hi JN,

I have now reinstalled the latest version but I am still getting everything in ‘Times'. From what you say, should now be in mono so not sure what is happening.

To make things ‘cleaner’ I did a fresh ‘opam install plotkicadsch’  in a completely new repo and I am on version 1.4 according to the install log.

The text positioning issue I mentioned previously had been resolved but the font issue unfortunately has not.

BW

John

jnavila, 2017-11-14:
Version 1.4 does not include the fix for the font, but the latest Git does. The SVG viewer might be picky about font specifications.

Gasman2014, 2017-11-14:
OK will have play with that (but I have got to  get my appraisal and tax return done :(

BW

John

jnavila, 2018-01-21:
I was able to reproduce the serif-font issue with Firefox. The latest version forces a sans-serif font which seems to be understood by Firefox. So it might also help for you.

Gasman2014, 2018-01-30:
Yes, now working. Also like your visual diff - hope that my KiDiff was helpful!

closed 2018-01-30 by Gasman2014

jnavila, 2018-01-30:
In fact, the screenshot of visual diff is from the new diffing strategy : diff at the level of drawing primitives and compute a diff in vector image. This allows to zoom at any level. You can try `plotgitsch` for two directories also (if it can be of use in the case of Fossil).