jnavila/plotkicadsch #4: Can't install versions which require kicadsch

Gasman2014, 2017-11-08:
I have previously managed to install with `opam install plotkicadsch`
Updating to any revisions more recent than [dc287e490c31ea9711ed08476f2c3964f770e63a]
are failing with the following message. I have tried an `opam update` and `opam upgrade` but without success. I have reverted back to [2e6c1eb5ff5082c59b0772f325ff38cf40bfee49] which is fine.

```
The following dependencies couldn't be met:
  - plotkicadsch -> kicadsch
Your request can't be satisfied:
  - No package named kicadsch found.

No solution found, exiting
```

jnavila, 2017-11-09:
I guess you tried to install from the git repo. In this case, you have to issue before:

```
$ cd plotkicadsch
$ opam pin add kicadsch .
$ opam install plotkicadsch
```

Anyway, I pushed plotkicadsch to opam. So, normally, you don't need to clone the repo to install `plotkicadsch` now. This command gets you the latest "stable" version (0.1.4), not the development one in this git repo.

I'm still fighting with image placement, but I should push a new version after.

jnavila, 2018-01-30:
Version 0.2.0 is now available from opam with plenty of new features !

closed 2018-01-30 by jnavila