jnavila/plotkicadsch #5: Installed with opam

leoheck, 2018-02-01:
I got this issue, trying to get the help as described in your README.md

```
➜ plotgitsch --help                            
Fatal error: exception Plotgitsch.InternalGitError("cannot parse rev --help")
```

Installed with opam, version
```
➜ opam install plotkicadsch
[NOTE] Package plotkicadsch is already installed (current version is 0.1.4).
```

leoheck, 2018-02-01:
I am trying to use it as follows:
```
plotgitsch -ifirefox
```
But I'm receiving the same error/issue.
Actually, I don't know how to use it.
Do you have any help?

jnavila, 2018-02-01:
Have you updated your opam? The latest version of plotgitsch in opam is 0.2.0.

leoheck, 2018-02-01:
Hmm, It was the first time using opam. I just installed it and upgraded wit upgrade command. I will try to upgrade something to get this 0.2.0 version installed.

jnavila, 2018-02-01:
With a new installation.... There's a dependency to ocaml's version in kicadsch. Can you try the following then:

```bash
$ opam switch 4.06.0
$ eval `opam config env`
$ opam update
$ opam install plotkicadsch
```

leoheck, 2018-02-01:
Great! It worked fine. It installed version 0.2.0

Now, just a minor issue... I can't get the version with the command bellow.
```
➜  plotgitsch --version
%‌%VERSION%%
```

leoheck, 2018-02-01:
I created the git-imgdiff script as you said once and I am having this Git Exception.
I am using the latest (nightly) Kicad. This can be an issue with plotgitsch

```
➜ plotgitsch -ifirefox
internal diff and show with firefox between Git rev HEAD and file system .
Git Exception: path not found: /
```
```
➜ plotgitsch 
use compare (ImageMagick) between bitmaps between Git rev HEAD and file system .
Git Exception: path not found: /
```

jnavila, 2018-02-02:
for `%%VERSION%%` I just realized that the version watermark does not correctly convert to the actual version. It's fixed in HEAD.

For the git exception, I can reproduce it if I don't run `plotgitsch` at the root of a git working copy containing the kicad project. How is your repo laid out?

leoheck, 2018-02-02:
My schematics are not on root folder. 
The schematics folder have the kicad project and all its files.

```
my_project
├── firmware
└── schematics
```

Running on root it gives me:
```
•100% ➜ plotgitsch
use compare (ImageMagick) between bitmaps between Git rev HEAD and file system .
```

Running on the schematics folder:
```
•100% ➜ plotgitsch   
use compare (ImageMagick) between bitmaps between Git rev HEAD and file system .
Git Exception: path not found: /
```

jnavila, 2018-02-02:
OK, that 's a bug due to the assumption.
The first command works correctly but does not find any schematic files to compare. The second one is the bug.
 I need to rework the project (hopefully during this week-end).

leoheck, 2018-02-02:
I see, probably you can still make these assumptions but you can keep it more flexible if you create an optional flag to pass the git-root-path and another one to tell which folder the schematics are...

Something like this, I guess...

```
# -g : git root folder ... we don't need to be in the project folder to run the script
# -d : schematics directory inside the root folder... schematics can be in a different folder...

plotgitsh -g /tmp/my_git_project -d path/to/schematics
```

leoheck, 2018-02-02:
I can give a look if you tell me the file that handles this command line.

jnavila, 2018-02-02:
I was starting to implement the recursive search up of git root and then recursive search down of *cache-lib and *.sch, but your proposition makes more sense if you have multiple projects in a single repo (it should be frown upon). I'll try to dig that way.

In the mean time, if you want to try out, it's all located in https://github.com/jnavila/plotkicadsch/blob/ad418afc49c44bbb50ad3100dd1452a2b581b258/plotkicadsch/src/plotgitsch.ml 

It's in Ocaml, which may be quite amazing to first comers, and it's spiced up by using first class modules. Let me know if you reach something interesting.

jnavila, 2018-02-02:
In fact, I would simplify your proposition, by requiring the command be launched at the root of the repo, but allowing to specify a relative path to diff.

leoheck, 2018-02-02:
Thank you for the references. I never heard about OCaml, so, I'm just starting slow, because it is very different than the things I already saw. 

I actually can compile your repository, I am using the master branch. I saw that the `%%VERSION%%` issue wasn´t fixed here. Am I doing something wrong? 

Right now, I am just starting to add something like this https://ocaml.org/learn/tutorials/command-line_arguments.html to the file you referenced me but I am struggling with the syntax of the OCaml :)

I also saw that `plotkicadsch` has a better command line to show help messages. IMHO this way is better than the way `plotgitsch` is doing. But if you are planning to add a man page, the way `plotgitsch` is doing is perfect.

```
•100% ➜ plotkicadsch --help
plotkicadsch prints Kicad sch files to svg
  -l specify component library
  -f sch file to process
  -help  Display this list of options
  --help  Display this list of options
```

jnavila, 2018-02-03:
The `%%VERSION%%` stuff is only useful when packaging and publishing to opam. So it's ok to see it when compiling by yourself.

I went for the [cmdliner](http://erratique.ch/software/cmdliner/doc/Cmdliner) because it's more powerful than the simple`Args` module and allows to mimic the argument handling of `git diff`.

Anyway, I pushed a branch `git_subdir`. The idea is simply to diff the current folder, but to work correctly, we have to find the git rootdir. As you already know how to compile it, can you try if it fixes your issue?

leoheck, 2018-02-03:
Cool, I could build it.

I ran it on my git root folder as you said, look the output:
```
➜ /home/lheck/Documents/plotkicadsch/_build/install/default/bin/plotgitsch 
use compare (ImageMagick) between bitmaps between Git rev HEAD and file system .
```

jnavila, 2018-02-03:
Hum, I didn't make myself clear: you have to run it in the kicad folder. The git root folder is automatically detected by recursing up and the relative path to it is also computed (needed for finding files in the git archives).

```bash
 $ cd schematics
 $ plotgitsch -ifirefox
```

leoheck, 2018-02-03:
I also did it. But I got an error related with /. I will run it again when
I reach my computer.

jnavila, 2018-02-03:
OK. I tested with one and two levels of depth for kicad projects and it works (for me). Would your repo be clonable by any chance?

BTW, to compile and test the project, the simplest is to issue:
```
 $ make build install
```
then `plotgitsch` should be installed in your path.

leoheck, 2018-02-03:
Thanks for the instructions on how to install it. It was easier installing it.

No, my repo cannot be shared, sorry. But I found [this](https://github.com/Galvant/gpibusb-pcb) repo that I could test. I got better results but I could not make it work yet due to the lack of libraries, I guess.
```
➜ plotgitsch -ifirefox 
internal diff and show with firefox between Git rev HEAD and file system .
Exception ("Kicadsch__Kicadlib.MakePainter(P).Component_Not_Found(\"CONN\")")
could not match component header (DEF usb-b:Conn Conn USB 0 0 N Y 1 F N): scanf: bad input at char number 20: looking for '0', found 'U'
```
*HOWEVER*, I just created [this](https://github.com/leoheck/plotgitsh-kicad-test) new repository to test it. 
Now, **I am seeing plotgitsch working**. The repository hierarchy is the same as mine repo. It also shows the differences between in any schematic sheets. It is very good. I have to figure it out why it is not working with my other repo.

jnavila, 2018-02-03:
Ah! At last some results! I was a bit concerned that you would just let this down out of despair. 🏁 

I'm going to investigate the failing repo because it seems to be yet another another assumption on component naming convention.

jnavila, 2018-02-07:
Hi,

Just a heads up on your issue. I tried to reproduce the error with the first repo you mentioned (gpibusb-pcb) but all I get is a missing `CONN` component (not included in the `-cache.lib`). What command did you use to experience the error that you are reporting ?

Did you manage to find the difference between your working repository and the test one?

leoheck, 2018-02-07:
No, not yet. I have plans to try again at the end of the day. If could ask
you something else, I also would like to have a good --help message if you
agree with that. Also, if you put some usage examples on README.md it will
help me and anyone's to put your repo working. Also, add the build
instructions on README.md. To make it more accessible for guys that do not
know mutch about software.

I will come back here with my results by the end of my day. I am on Brazil.
Thank you @jnavila

jnavila, 2018-02-07:
I guess the help message is already a lot better than what I usually commit... :wink: . The format of help message can be extended but it should remain lean. I would prefer a standalone guide document on how to use the tool, the `--help` being only a reminder of the available options of the command line.

For the README file, I already wanted to add more thorough installation instructions, but you're welcome to propose something that would help make it accessible to less techies (I don't qualify to accessibility expert).

jnavila, 2018-09-27:
Can we close this issue?

leoheck, 2018-09-28:
Yes, sure!

closed 2018-09-28 by jnavila