jnavila/plotkicadsch #7: Cant run it.

leoheck, 2018-08-24:
I cant use it. 
When I try to run, I get this message:

```
•100% ➜ plotgitsch       
use compare (ImageMagick) between bitmaps between Git rev HEAD and file system .
```
```
•100% ➜ plotgitsch --version
v0.3.0
```

leoheck, 2018-08-24:
The readme files should have at least one or two examples on how to run it.

```
•100% ➜ plotgitsch .
plotgitsch: internal error, uncaught exception:
            (Invalid_argument "pos + len past end: 0 + 4 > 1")
            Raised at file "pervasives.ml", line 33, characters 20-45
            Called from file "src/string.ml", line 127, characters 2-76
            Called from file "plotkicadsch/src/plotgitsch.ml", line 329, characters 5-31
            Called from file "plotkicadsch/src/plotgitsch.ml", line 336, characters 39-49
            Called from file "src/cmdliner_arg.ml", line 35, characters 22-29
            Called from file "src/cmdliner_arg.ml", line 214, characters 38-45
            Called from file "src/cmdliner_arg.ml", line 226, characters 14-41
            Called from file "src/cmdliner_term.ml", line 25, characters 12-19
            Called from file "src/cmdliner_term.ml", line 22, characters 21-30
            Called from file "src/cmdliner_term.ml", line 22, characters 21-30
            Called from file "src/cmdliner.ml", line 106, characters 32-39
```
```
•100% ➜ plotgitsch HEAD .
plotgitsch: internal error, uncaught exception:
            (Invalid_argument "pos + len past end: 0 + 4 > 1")
            Raised at file "pervasives.ml", line 33, characters 20-45
            Called from file "src/string.ml", line 127, characters 2-76
            Called from file "plotkicadsch/src/plotgitsch.ml", line 329, characters 5-31
            Called from file "plotkicadsch/src/plotgitsch.ml", line 336, characters 39-49
            Called from file "src/cmdliner_arg.ml", line 35, characters 22-29
            Called from file "src/cmdliner_arg.ml", line 214, characters 38-45
            Called from file "src/cmdliner_arg.ml", line 226, characters 14-41
            Called from file "src/cmdliner_term.ml", line 25, characters 12-19
            Called from file "src/cmdliner_term.ml", line 22, characters 21-30
            Called from file "src/cmdliner.ml", line 106, characters 32-39
```

leoheck, 2018-08-24:
```
•100% ➜ plotkicadsch -f circuit.sch 
Fatal error: exception Kicadsch__Kicadlib.MakePainter(P).Component_Not_Found("MyLib:Fiducial")
```

jnavila, 2018-08-24:
This project deserves more love. You can plot a specific sch file like this:
```
 $ cd circuit
 $ plotkicadsch -l circuit-cache.lib  -f circuit.sch
```
This should create a file `circuit.svg` from the schematic file.

As for plotgitsch:
```
 $ cd circuit
 $ plotgitsch -ifirefox HEAD^
```

jnavila, 2018-08-24:
I just checked, the exception you get is due to a bug in arguments parsing. If the argument is a directory, you have to prefix it with "dir:", or if it's ".", you can simply omit it, otherwise it's a git ref (which must be longer than 4 chars).

I'll fix this bug shortly.

jnavila, 2018-08-24:
It's any library (or libraries, you can specify multiple -l <lib>), in order to get all the needed symbols for the current schematic.

If you don't have the <project>-cache.lib, I'm sorry that plotgitsch won't work at all.

leoheck, 2018-08-24:
Oh, really? It is good to know. So, do you know a way to generate it again?

You should add all of this to the Readme. :)

jnavila, 2018-08-24:
At least, plotgitsch works correctly with your exercise project `plotgitsh-kicad-test`.

jnavila, 2018-08-24:
From what I know, it's important to keep the "-cache.lib" file committed with your project, because otherwise, your project history may be relying on lib files which have evolved since it was written. You really want to check out your project and be able to open it, see it,  in the state it was when you committed.

leoheck, 2018-08-24:
I couldn't make it work with my plotgitsh-kicad-test now.

I didn't know that. I am looking for a way to generate the file again

leoheck, 2018-08-24:
I mean, `plotgitsch` opens the imagemagic, should I do something, or analyze the image with it? I thought that it was going to generate a file.

It is pretty hard to use imagemagic, haha.

jnavila, 2018-08-24:
you can use the -i<browser command> option, with your favorite browser i.e. `-ifirefox` for instance. I no longer use imagemagick, because the diff are less informative. My projects are well behaved with the tool I developed :laughing:

leoheck, 2018-08-24:
Ok, regenerating cache lib is easy.
https://forum.kicad.info/t/missing-cache-lib-in-win7/6817/2

Cool, is it open source? Can you share it or a screenshot?

Look, this, may you can help :1st_place_medal: 
https://bugs.launchpad.net/bugs/1788913

jnavila, 2018-08-24:
for your project:
```
 $ cd circuit
 $ plotgitsch -ifirefox HEAD~2 HEAD~4
```
I get:
[image]

leoheck, 2018-08-24:
Nice, was able to recreate it here. What if you add a option to cropt the image reducing the whitespace?

jnavila, 2018-08-24:
I'm working on an interactive web page which would allow to navigate between diff zones. But I'm quite slow...

leoheck, 2018-08-24:
I couldnt add library option with git, is it possible?

```
plotgitsch -ifirefox HEAD HEAD~1  
internal diff and show with firefox between Git rev HEAD and Git rev HEAD~1
Exception ("Kicadsch__Kicadlib.MakePainter(P).Component_Not_Found(\"DeepX:Fiducial\")")
```

jnavila, 2018-08-24:
No, it isn't right now. I didn't anticipate this request.

leoheck, 2018-08-24:
This is the same as https://github.com/jnavila/plotkicadsch/issues/6

jnavila, 2018-08-24:
It shouldn't be too complex.

jnavila, 2018-08-24:
But, again, I insist that it's saner to commit the cache library with the project. If your history is not too long, you can rewrite it and add the -cache.lib file from the beginning.

leoheck, 2018-08-24:
I didn't understand. I recreated the -cache.lib. Do I have to put it on git to make it work?

jnavila, 2018-08-24:
yes, it must appear in the git commits, just like for `plotgitsh-kicad-test`.

leoheck, 2018-08-24:
Is it important to add it from the beginning?

jnavila, 2018-08-24:
Well, every commit where it doesn't appear cannot be used for a diff. `plotgitsch` uses the `-cache.lib` file available in the refered commit to perform the drawing.

leoheck, 2018-08-24:
Alright! Now I understand the process, thanks @jnavila I will try to use it for my next commits.

jnavila, 2018-08-24:
Thanks for asking questions. That makes the meat of the user's guide (to come) :wink:

leoheck, 2018-08-24:
COOL! It is working. It was sooner than I expected.

jnavila, 2018-08-24:
beware though: this is still early alpha.

leoheck, 2018-08-24:
Alright! I will use it to help in the reviewing process because currently is a little bit complicated to use it with git.

jnavila, 2018-08-24:
Did you compile using opam or from the repo?

leoheck, 2018-08-24:
I dont know. This opam thing was hard. I thought that I had to clone the repo and then execute opam to build it. So, probably I was using opam.

leoheck, 2018-08-25:
Oh, great! I will test it.
But how I compile it myself? Is it using opam too?
Because I was confusing about that. Opam is a package manager ou a compiler, maybe both?

jnavila, 2018-08-25:
I guess you used opam, at least to get the compiler and the dependencies. Then, 

- either you installed plotkicadsch from opam, which means you requested opam to install plotkicadsch and let it  download the sources, compile them and install the binaries
-  or you used this repo i.e. you cloned it, tagged it as the source for plotkicadsch in opam and installed  the same way as previously, but opam used your local repo.

In the second case, it is quite easy to:
1. fetch the branch `git fetch origin`
2. check it out,`git checkout origin/specify_ext_lib`
3. run `opam update` to update the list of new packages
4. run `opam upgrade` to install the new version

leoheck, 2018-08-28:
Cool, I was able to upgrade it (and learn opam a litle bit, thanks).
Everything is working fine
Also, I was going to ask this for you, but I was able to open all the sheets with changes at once. 
That is great.

I am using it with:
```
plotgitsch -l circuit-cache.lib -inkscape HEAD^ HEAD
```

jnavila, 2018-08-28:
Great :tada:

closed 2018-08-28 by jnavila