jnavila/plotkicadsch #10: Some CLI flags to control the output

leoheck, 2018-08-28:
@jnavila, what if you add some command line flags to control the output better?

Is something hard to do?

For example, something to control:
- the line width (to make everything more visible)
- the background color
- the color of the left file
- the color of the right file

jnavila, 2018-08-28:
It isn't high on my priority list. I'd like to focus on interactivity and add these options in the browser.

jnavila, 2018-08-29:
I can already rework the drawing to make it easier to read.

jnavila, 2018-08-29:
The line width can be changed. Which colors would you expect?

leoheck, 2018-08-29:
I was thinking to give the flexibility for the user exposing a flag to set
the RGB. I can't see much difference when it has red and green :) including
when the line is thin.

I can also control everything with inkscape but if the tool gives this
flexibility it is better.

Also, a flag to keep the svg files instead of removing them is good too.

jnavila, 2018-08-29:
If you have time, can you test the branch enhance_svg?

```console
 $ git fetch origin
 $ git checkout -b enhance_svg origin/enhance_svg
```

leoheck, 2018-08-29:
Sure! What should I test? I mean, which enhancements should I find?
- The line width is better.
- The colors are darker now but it is even harder for me to distinguish. Previous colors were better.

Anything else?
Is it hard to pass this from the command line? I was looking at the code but as I never saw this language it is pretty much hard to understand where to add this. Of course, I need to study it more. :)

jnavila, 2018-08-29:
It is hard to pass the colors to the svgPainter. The code needs to be refactored to have supplementary parameters. For the moment, the thicker line width is kept but the colors are rolled back.

jnavila, 2018-08-29:
It's merged in master. I'll have to think of an elegant way to pass colors.

jnavila, 2018-09-07:
see #16

PR #16 (2018-09-07): Param svg painter
@leoheck

This should fix #10 . In fact, it wasn't as disruptive as anticipated, because my first analysis of the feature was wrong: the feature is not fully in the painter, but also partly in the differ object.

So in this branch you can test the feature with e.g.:
 ```console
 $  plotgitsch -ifirefox --colors=FF0000:0000FF:FFFFFF:000000
```

The colors field is the RGB values of old, new, foreground and background colors. 
To test, as usual, in `plotkicadsch` sources:

```
 $ git fetch  git@github.com:jnavila/plotkicadsch.git param_svgPainter
 $ git checkout FETCH_HEAD
 $ m [... cut]
closed 2018-09-07 by jnavila