jnavila/plotkicadsch #12: provide binaries

jnavila, 2018-08-31:
If possible provide installers for
 * linux: should be easy, but less priority
 * mac: seems doable, ocaml is used on Mac, have to check how dmg are done
 * windows: hard point, but that's where the users base is

see [0install](https://github.com/0install/0install) for ideas?

leoheck, 2018-09-07:
Did you already try to install it on windows? My colleges are tied to windows =|
So, if you already tried let me know, maybe we can work together on that.
I will give a look on 0install too.

jnavila, 2018-09-07:
I started making some tests. The binaries compile, but I haven't found yet the way to run git as a subprocess for now. But it's not lost yet.

leoheck, 2018-09-07:
Git should be on PATH. It should be the same as running it in Linux.

jnavila, 2018-09-07:
I tried, even in git bash, but couldn't make it work. You can test the binary output in case I overlooked anything:
https://ci.appveyor.com/api/projects/jnavila/plotkicadsch/artifacts/opam/4.06.0+mingw64c/bin/plotgitsch.exe

leoheck, 2018-09-10:
It looks promising, but couldn't launch a program to visualize the images.
I was trying a full path to chrome.exe
Is it right?

jnavila, 2018-09-10:
Normally, you needn't provide the full path. It should work with just `-ichrome`. Anyway, I'll check if I can correctly quote the parameters for Windows, but this seems quite obscure.

leoheck, 2018-09-10:
Good! It is working as you said with "chrome". 

Well, considering on Linux, it should work if the binary is located with the PATH variable. So, when the binary is not located on the PATH variable, its full-path should be given (should work)

Since I couldn't launch the chrome.exe (or chrome) from the command line, I thought to add the whole path there.

jnavila, 2018-09-10:
In fact, Windows provides a resolver: `start`; so launching `start chrome` works, and that's just how I've done it.

leoheck, 2018-09-10:
Hm that is Interesting.

jnavila, 2018-09-10:
I'm not still sure whether the indirect paths are correctly managed (project as a subdir of git working copy). And some form of packaging is still needed. :construction_worker_man:

leoheck, 2018-09-10:
Sure, the packaging is definitely a thing for windows users.
But I couldn't understand why you think indirect paths are correctly managed.
Give me an example, so I can try to recreate it.

jnavila, 2018-09-10:
That's the case for your project where the kicad project is a subdir of the main git project. I may have messed up with path separators.

jnavila, 2018-09-10:
Another issue is that the files generated for the diffs cannot be removed under windows as long as they are opened in the browser. So the keep option is useless.

leoheck, 2018-09-10:
Actually, I was not testing it on a Kicad project inside a git sub dir. 

Maybe it is better to generate the diff files on a temp folder. So the keep flag should just copy them to the projects dir.

jnavila, 2018-09-10:
Yeah, that would be better; the good news is that you managed to reproduce a working setup. Thanks for that.

jnavila, 2018-09-25:
Windows binaries are available here for each tag:
https://github.com/jnavila/plotkicadsch/releases

Next tag should be v0.4.0

closed 2018-09-25 by jnavila

gbip, 2018-11-23:
Could you include linux binaries ? I'd like to integrate your tool with a github bot which runs on linux.

gbip, 2018-11-23:
I think that expecting Kicad users to be expert in ocaml build system is requiring too much : installing opam is not that simple for everyone and the build process might fail because of issues unrelated to your project...

follower, 2019-08-16:
As I'm currently waiting for `Processing  4/12: [ocaml-base-compiler: make world.opt]` to complete I'll take the time to add my own request for other platform binaries. :) (Primarily Linux in my case.)

Needing to install `opam` (which asks questions & wants to modify profile scripts & environment variables) and then wait for it to (I assume) install an entire OCaml compiler & environment is quite a barrier to trying out `plotkicadsch`.

Given that OCaml seems capable of producing standalone binaries I'd encourage you to consider producing binaries for Linux & Mac also (as `opam` itself does :) ).

Thanks for your work on plotkicadsch, I look forward to trying it out.

follower, 2019-08-16:
Back again cos I'm waiting for `Processing 4/12: [ocaml-base-compiler: make world.opt]` (again!) because apparently the default version installed is now `4.08.1` which didn't work with `plotkicadsch`.

With version `4.08.1` the `[...] install plotkicadsch` command fails with "dependencies couldn't be met" with a long list.

In addition the command `[...] switch 4.06.0` fails and asks if I meant  `[...] switch create 4.06.0`. Which then triggers another complete compiler + environment install... (On which I'm currently waiting for completion.)

_time passes_

When I try to install in `4.06.0` switch with ``[...] install plotkicadsch` it starts to install the dependencies and then fails with:
```
[...snip...]
# Fatal error: exception Out of memory

<><> Error report <><><><><><><><><><><><><><><><><><><><><><><><><><><><><><><>
┌─ The following actions failed
│ λ build dune 1.11.1
└─ 
[...snip...]
```

Which I'm guessing might be related to the fact I'm doing all this in a VM?

Anyway, I think this is probably my limit in trying out `plotkicadsch` at this point but hopefully it serves as a data point toward why additional binaries could be helpful.

Thanks for your time.

follower, 2019-08-16:
Another update as I couldn't leave well enough alone, so... :D

I closed a few browser tabs in the VM and tried re-running the install which proceeded further but then stop due to running out of disk space while building `core_kernel`. (Apparently ~3GB free wasn't enough for the install.)

I cleared a little more space and tried the install one last time and it completed. :)

And then I was able to perform a test schematic conversion. :)

jnavila, 2019-08-16:
@follower glad you haven't given up. I read you are using a VM. Is your Kicad under a VM? Why not work native on your OS?