jnavila/plotkicadsch #15: Is it working?

leoheck, 2018-08-31:
I am running the upgraded master, but I can't make it work again.
Probably I am doing something wrong again.

```
deepx/BIRDCALL_HOMEBASE/homebase-hardware on  homebase-0.1-initial-layout [$?] 
•100% ➜ plotgitsch -l homebase-cache.lib -ifirefox HEAD HEAD^1
plotgitsch: unknown option `-l'.
Usage: plotgitsch [OPTION]... [FROM_REF] [TO_REF]
Try `plotgitsch --help' for more information.
```

leoheck, 2018-08-31:
It was not upgraded, =/ sorry.

closed 2018-08-31 by leoheck

jnavila, 2018-08-31:
Please note there is also the new --keep option to prevent deletion of diff files

leoheck, 2018-08-31:
Is this thing help to create a good command line parser with integrated help?
Look the output on the 3rd box. 

[image]

Found it here:
https://rosettacode.org/wiki/Command-line_arguments#OCaml

jnavila, 2018-09-01:
You can discover the command line options with `plotgitsch --help`, I used `Arg` for `plotkicadsch`, but it is a lot less powerful than `Cmdliner` which is used in `plotgitsch`, particularly for managing variable number of arguments.

I still don't clearly get what's the issue with how the help page is shown with `plotgitsch`, I find it well organized.