jnavila/plotkicadsch #17: Opening with firefox

leoheck, 2018-09-03:
I have a project with 5 schematics sheets and 3 of them with changes. I am using firefox to display the diffs. Do you know why 2 of the diffs open in one firefox window (different tabs) while the last one is opening on a new window? It is working fine with chrome.

jnavila, 2018-09-04:
It's a wild guess: plotgitsch fires up multiple times "firefox <result>.svg". This exec spawns a firefox process which looks for an already opened firefox window (maybe serving on a unix socket). If not found, the firefox process opens a new window; otherwise it opens a new tab in the found running instance. If the firefox processes are exec'ed to quickly in a raw, the server does not reply to some clients in time and the clients in failure generate a new window.

Chrome seems to be able to cope with the rate of client requests for new tabs.

leoheck, 2018-09-04:
Cool, I was thinking the same.

closed 2018-09-04 by leoheck