jnavila/plotkicadsch #22: Component not found issue

MitjaNemec, 2018-11-03:
Hi!

First of all thanks for the windows binaries. I appreciate the effort.

I can not get the program to work. Even when I specify correct -cache.lib file, I get the `Component_Not_Found` error. The error message says it is looking for symbol in `library:symbol`, but under 5.0.1 in cache file, the symbol is stored as `library_symbol`.

I know that the cache file format changed before V5, and I think also once V5 was already released, but I am not certain. Could this be an issue?

jnavila, 2018-11-03:
Oh. You're right. They changed it. Guess I'll have to introduce a new revision.

shrub77, 2018-11-04:
@jnavila This also breaks the kicad_playground project

jnavila, 2018-11-04:
@shrub77 I just tested again with kicad_playground, without opening the project in kicad and the user's guide works as intended.

Where are you seeing an issue?

shrub77, 2018-11-04:
@jnavila I see you just made a commit - did you try with the new build? I was using the one from yesterday. How can I update my version? I installed with opam

jnavila, 2018-11-04:
@MitjaNemec For Windows builds, you can get the latest binaries from Appveyor at

https://ci.appveyor.com/api/projects/jnavila/potkicadsch/artifacts/opam/ocaml-variants.4.06.0+mingw64c/bin/plotgitsch.exe

shrub77, 2018-11-04:
@jnavila i did as you advised, but cannot compile the repo - dune build returns a linking error. I am not familiar with ocaml or dune, not sure how to proceed. Do you have built binaries for the latest release like you do for windows?

shrub77, 2018-11-04:
@jnavila I managed to solve this problem by fixing some dependencies I was missing. I can now proceed with the tutorial, thanks

MitjaNemec, 2018-11-06:
@jnavila thanks for the binary, but I don't have access as I get the following message on appveyor:
`{"message":"Project not found or access denied."}`

jnavila, 2018-11-06:
Sorry, there was a typo:

https://ci.appveyor.com/api/projects/jnavila/plotkicadsch/artifacts/opam/ocaml-variants.4.06.0+mingw64c/bin/plotgitsch.exe

MitjaNemec, 2018-11-16:
When trying the new version on playgroudn project I get a following error:
```
D:\Mitja\Plate\kicad_playground\schematic>plotgitsch -k
use compare (ImageMagick) between bitmaps between Git rev HEAD and file system .

Exception (Invalid_argument "index out of bounds")
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '# Device_LED
'hrowing away line '#
'hrowing away line 'DEF Device_LED D 0 40 N N 1 F N
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '# Device_R
'hrowing away line '#
'hrowing away line 'DEF Device_R R 0 0 N Y 1 F N
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '# Timer_LM555
'hrowing away line '#
'hrowing away line 'DEF Timer_LM555 U 0 20 Y Y 1 F N
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '# power_+5V
'hrowing away line '#
'hrowing away line 'DEF power_+5V #PWR 0 0 Y Y 1 F P
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '# power_GND
'hrowing away line '#
'hrowing away line 'DEF power_GND #PWR 0 0 Y Y 1 F P
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '#End Library

D:\Mitja\Plate\kicad_playground\schematic>
```

jnavila, 2018-11-16:
@MitjaNemec 

Are you sure you grabbed the latest version (0.4.1)? You need to run the commands in the git bash environment, not under command.com.

The error message is quite strange also, because the format of the library seems to be post v5.x, whereas the kicad_playground is normally in version v.4. Did you convert the project to v5?

MitjaNemec, 2018-11-28:
Closing as it was fixed. I can not close #23, as I did not open it (from the Github perpective)

closed 2018-11-28 by MitjaNemec