jnavila/plotkicadsch #23: When trying the new version on playgroudn project I get a following error:

jnavila, 2018-11-16:
```
D:\Mitja\Plate\kicad_playground\schematic>plotgitsch -k
use compare (ImageMagick) between bitmaps between Git rev HEAD and file system .

Exception (Invalid_argument "index out of bounds")
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '# Device_LED
'hrowing away line '#
'hrowing away line 'DEF Device_LED D 0 40 N N 1 F N
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '# Device_R
'hrowing away line '#
'hrowing away line 'DEF Device_R R 0 0 N Y 1 F N
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '# Timer_LM555
'hrowing away line '#
'hrowing away line 'DEF Timer_LM555 U 0 20 Y Y 1 F N
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '# power_+5V
'hrowing away line '#
'hrowing away line 'DEF power_+5V #PWR 0 0 Y Y 1 F P
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '# power_GND
'hrowing away line '#
'hrowing away line 'DEF power_GND #PWR 0 0 Y Y 1 F P
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '#End Library

D:\Mitja\Plate\kicad_playground\schematic>
```

_Originally posted by @MitjaNemec in https://github.com/jnavila/plotkicadsch/issues/22#issuecomment-439329980_

jnavila, 2018-11-16:
Are you sure you grabbed the latest version (0.4.1)? You need to run the commands in the git bash environment, not under command.com.

jnavila, 2018-11-16:
The error message is quite strange also, because the format of the library seems to be post v5.x, whereas the kicad_playground is normally in version v.4. Did you convert the project to v5?

MitjaNemec, 2018-11-16:
> Are you sure you grabbed the latest version (0.4.1)? You need to run the commands in the git bash environment, not under command.com.

No I used the one you pointe me to in #22. Sorry for the confusion I'll try the new one. What is the issue, when running it from command.com? I'll go and run it within git bash, but if it can run within command.com, then it would be possible to integrate it as a diff viewer

> the error message is quite strange also, because the format of the library seems to be post v5.x, whereas the kicad_playground is normally in version v.4. Did you convert the project to v5?

Yup, I've opened changed and saved the project with V5.0.1

jnavila, 2018-11-16:
>What is the issue, when running it from command.com? I'll go and run it within git bash, but if it can run within command.com, then it would be possible to integrate it as a diff viewer

TBH, I haven't tested much under windows. At least, I know that git bash reproduces most my native environment. 

What I don't know is whether the call to external tools scripts is supported. You are using the command without the `-i` option, which triggers the `git-imgdiff` script which may not work under windows.

MitjaNemec, 2018-11-16:
I get the same with 0.4.1 regardless where I run it from (thought to run it from cmd, you have to append the path to it to system enviroment path variable.

But I can get the `plotkicadsch` to generate `.svg` file, which is something at least, and with imagemagic, I should be able to generate diff and show it.

jnavila, 2018-11-16:
I just tried again, after opening and modifying the project with kicad 5.0.1 and it still works. I don't understand what you are facing. I guess I'll have to find a Windows to make some more tests on the platform. What command did you type? What is the output with 0.4.1?

MitjaNemec, 2018-11-19:
So running `plotkicadsch -f LEDs.sch -l kicad_playground-cache.lib` I get proper `.svg` file. same goes for `plotkicadsch -f kicad_playground.sch -l kicad_playground-cache.lib`

Running `plotgitsch --version ` I get `%%VERSION%%` so there is another bug, but it is really a nuisance.

Running `plotgitsch -k` or `plotgitsch -t` or `plotgitsch -k -l kicad_playground-cache.lib` or `plotgitsch -t -l kicad_playground-cache.lib` or `plotgitsch -i firefox.exe -l kicad_playground-cache.lib` I get the error mentioned above.

Hope this helps a bit. If you need additional testing on Windows I am always glad to help

jnavila, 2018-11-19:
The bug for `plotgitsch --version` is due to the fact that the binary is not compiled for a release on opam. I'll try to have a correct version outside opam.

`plotgitsch` doesn't accept the same options as `plotkicadsch`. Can you send me your modified `kicad_playground-cache.lib`? I guess there's something that breaks my app in it.

jnavila, 2018-11-20:
I just reproduced your output with your cache-lib, on Linux! Time to debug. Thanks.

jnavila, 2018-11-20:
This is due to a change in line endings in the lib file. The one you privided me is DOS line endings, whereas the standard one is UNIX line endings. Have you tried to open the file in a text editor?

jnavila, 2018-11-20:
Will do. It's safer anyway.

MitjaNemec, 2018-11-20:
Obviously I've also got DOS line endings in .sch files

MitjaNemec, 2018-11-28:
Thanks for fixing it. Tested on windows and it is working also from command line. As I don't know what is the etiquette I have to ask who is supposed to close this issue and #22? Me or you?

jnavila, 2018-11-28:
If you're ok with what was done, please close it. That's usually how it goes.

closed 2018-11-28 by jnavila