jnavila/plotkicadsch #24: Add option to specify output folder/filename

MitjaNemec, 2018-12-03:
Hi!

As I am get closer to integrating `plotkicadsch` into a complete diff tool for TortoiseGit (Windows git GUI client), I'd like to have an option to specify output folder or even better full filename.

Thanks

jnavila, 2018-12-03:
Something I don't understand: isn't plotgitsch working for you?
Now, for the option, it's `plotkicadsch` you are refering to. How do you intend to use it?

MitjaNemec, 2018-12-03:
plotgitsch is working, but the results is a diff of the whole project. As I am used to checking differences file by file using TortoiseGit, I wanted to use plotkicadsch as a svg generator for different revisions of a file in question and then generate diff and view it. So the git part of the system is handled by TortoiseGit. As such integration of plotgitsch into TortoiseGit is not really possible.

It all comes down to how do you work with git. And under Windows I am used to TortoisGit and the above way is the easiest to integrate.

jnavila, 2018-12-04:
OK. I'm going to add an option to `plotkicadsch` for target directory, but I think I can also add an option to `plotgitsch` to restrict to one file.

MitjaNemec, 2018-12-04:
Great, thanks. 

As for modifying `plotgitsch` the TortoiseGit is passing both revisions of the file to make the diff [1], if you could add an option to `plotgitsch ` to just make the diff of two supplied .sch files, this would be even better as I would get rid of the imagemagick dependency. But in this case, there is no need for git in `plotgitsch`.

Ideally what I would need is a .sch diff tool, which takes two .sch files as an input and produces diff in .svg format as an output, with option to specify both input files and an output file path. But I can make do with `plotkicadsch`.

[1]. [TortoiseGit using external diff tools](https://tortoisegit.org/docs/tortoisegit/tgit-dug-settings.html#tgit-dug-settings-progs)

jnavila, 2018-12-08:
Version v0.4.3 brings the following:

- `plotkicadsch` now accept a `-o` option for setting up output directory
- `plotgitsch` now accept the `-f <filename>` or `--file=<filename>` to restrict diff to only one file

MitjaNemec, 2018-12-08:
Almost there. There is a regression in `plotkicadsch` when file path is specified with absolute path. Note that library can be specified with absolute path. This was not an issue in 0.4.2
```
D:\Mitja\Plate\kicad_playground\schematic>plotkicadsch -f kicad_playground.sch -l kicad_playground-cache.lib

D:\Mitja\Plate\kicad_playground\schematic>d:

D:\Mitja\Plate\kicad_playground\schematic>cd ..

D:\Mitja\Plate\kicad_playground>cd ..

D:\Mitja\Plate>cd ..

D:\Mitja>cd ..

D:\>plotkicadsch -f d:\Mitja\Plate\kicad_playground\schematic\kicad_playground.sch -l D:\Mitja\Plate\kicad_playground\schematic\kicad_playground-cache.lib
Fatal error: exception Unix.Unix_error(Unix.EUNKNOWNERR -123, "open", "./d:\\Mitja\\Plate\\kicad_playground\\schematic\\kicad_playground.svg")

D:\>cd Mitja\Plate\kicad_playground\schematic

D:\Mitja\Plate\kicad_playground\schematic>plotkicadsch -f d:\Mitja\Plate\kicad_playground\schematic\kicad_playground.sch -l kicad_playground-cache.lib
Fatal error: exception Unix.Unix_error(Unix.EUNKNOWNERR -123, "open", "./d:\\Mitja\\Plate\\kicad_playground\\schematic\\kicad_playground.svg")

D:\Mitja\Plate\kicad_playground\schematic>plotkicadsch -f kicad_playground.sch -l D:\Mitja\Plate\kicad_playground\schematic\kicad_playground-cache.lib

D:\Mitja\Plate\kicad_playground\schematic>
```

jnavila, 2018-12-08:
oups... This part was brittle. Will fix shortly.

MitjaNemec, 2018-12-09:
Works. Though if I dare to nitpick, it still crashes when the folder path ends with backslash. E.g.
`-o "C:\Users\MitjaN\AppData\Local\Temp\TortoiseGit\"` crashes
`-o "C:\Users\MitjaN\AppData\Local\Temp\TortoiseGit"` works