jnavila/plotkicadsch #25: Component not found

jagjordi, 2019-04-07:
When runing the plotgitsch the following error emerges `Exception ("Kicadsch__Kicadlib.MakePainter(P).Component_Not_Found(\"Device:D_Schottky\")")`

I installed the last release version and the error persists.
I attach the project that is giving me the error.

[en.zip](https://github.com/jnavila/plotkicadsch/files/3051560/en.zip)

jnavila, 2019-04-07:
The file  `eV16-BSPD-cache.lib` was not commited. Please follow the advices at   https://jnavila.github.io/plotkicadsch/ for setting up your kicad project correctly.

jnavila, 2019-04-07:
I'm seeing in your `.gitignore` that you are ignoring precious files. I don't know where this .gitignore file comes from but it's utterly wrong.

jagjordi, 2019-04-07:
Thank for your reply. After using your `.gitignore` everything works as intended. As per your question, this gitignore is the one that github provides in their [gitignore](https://github.com/github/gitignore/blob/master/KiCad.gitignore) repo.

closed 2019-04-07 by jagjordi

jnavila, 2019-04-07:
Ah, thanks for the origin of the file; guess it's time to open a PR then :wink: