jnavila/plotkicadsch #26: Exception (Invalid_argument "Negative position: -1")

geekbozu, 2019-05-06:
Seems like there may have been a file format change on kicad 5.1.0?

I committed a blank schematic, Added a resistor, His save then ran 
```plotgitsch -ifirefox```
and this is the result. 
```Exception (Invalid_argument "Negative position: -1")```
seems to be tied to the part orientation.

geekbozu, 2019-05-06:
So I made some more commits and decided to try it again and it works now? I'm going to close this but it seems like when there are No components on the schematic and it tries to compare against it it fails.

jnavila, 2019-05-07:
Ah. I've never tried to compare schematics with no components.

jnavila, 2019-05-11:
Thanks for opening this issue. This should be fixed now.

closed 2019-05-11 by jnavila

geekbozu, 2019-05-11:
Sorry for not getting a demo up for you. Got busy soon after I made this :/  but thanks for fixing it! I'll give it a test when I get a chance.