jnavila/plotkicadsch #27: scanf: bad input at char number 11: character 'S' is not a decimal digit

arikrupnik, 2019-07-26:
I installed using opam on Ubuntu 19.04, if it matters. It appears that the relevant `scanf` chokes on an escaped quote mark. The offending character is `\"` which represents inches in my text. If I remove the quote mark, everything works. With the value as it is, I get this error:

> could not match Component F (F 1 "1/8\" SJ1-3544" H 10350 800 50  0000 R CNN): scanf: bad input at char number 11: character 'S' is not a decimal digit

The relevant component in my .sch is this:

```
$Comp
L dish-schematic-components:SJ1-3544 J1
U 1 1 5C87F705
P 10050 1200
F 0 "J1" H 10100 700 50  0000 R CNN
F 1 "1/8\" SJ1-3544" H 10350 800 50  0000 R CNN
F 2 "dish-components:CUI_SJ1-3544" H 10050 1200 50  0001 C CNN
F 3 "https://www.cui.com/product/resource/sj1-354x.pdf" H 10050 1200 50  0001 C CNN
F 4 "CP1-3544-ND" H 10050 1200 50  0001 C CNN "DigiKey"
	1    10050 1200
	-1   0    0    1   
$EndComp
```

jnavila, 2019-07-26:
Indeed, `scanf` does not interpret the `\"` sequence as an escape and the rest to the parsing fails. There may be ways to make is smarter. For now, escaped characters are forbidden. As a workaround you can use unicode characters like ″ which are not interpreted as quotes but render similarly.

jnavila, 2019-07-26:
OK, I found the bug. Can you compile and test from this repo?

arikrupnik, 2019-07-28:
Thank you for looking into this. I have no experience with OCaml, and I"m afraid I failed to build the GIT version.

I'm attaching the full output of how I tried to follow your directions. I'm probably missing something really small, and I feel embarrassed about it.

```
ari@omen:~$ git clone https://github.com/jnavila/plotkicadsch
Cloning into 'plotkicadsch'...
remote: Enumerating objects: 73, done.
remote: Counting objects: 100% (73/73), done.
remote: Compressing objects: 100% (38/38), done.
remote: Total 1240 (delta 26), reused 66 (delta 22), pack-reused 1167
Receiving objects: 100% (1240/1240), 345.96 KiB | 2.75 MiB/s, done.
Resolving deltas: 100% (718/718), done.
ari@omen:~$ cd plotkicadsch/
ari@omen:~/plotkicadsch$ opam pin add kicadsch .
[NOTE] Package kicadsch is currently pinned to git+file:///home/ari/global-sync/dish/pcb#Test-Fixture (version 0.5.0).
[kicadsch.0.5.0] synchronised from git+file:///home/ari/plotkicadsch#master
kicadsch is now pinned to git+file:///home/ari/plotkicadsch#master (version 0.5.0)

The following actions will be performed:
  ⊘ remove plotkicadsch 0.5.1
  ⊘ remove kicadsch     0.5.0*
===== ⊘ 2 =====
Do you want to continue? [Y/n] y

<><> Processing actions <><><><><><><><><><><><><><><><><><><><><><><><><><><><>
⊘ removed   plotkicadsch.0.5.1
⊘ removed   kicadsch.0.5.0
Done.
ari@omen:~/plotkicadsch$ opam pin add plotkicadsch .
[plotkicadsch.0.5.1] synchronised from git+file:///home/ari/plotkicadsch#master
plotkicadsch is now pinned to git+file:///home/ari/plotkicadsch#master (version 0.5.1)

Sorry, no solution found: there seems to be a problem with your request.

[NOTE] Pinning command successful, but your installed packages may be out of sync.
ari@omen:~/plotkicadsch$ opam update

<><> Updating package repositories ><><><><><><><><><><><><><><><><><><><><><><>
[default] synchronised from https://opam.ocaml.org

<><> Synchronising development packages <><><><><><><><><><><><><><><><><><><><>
[kicadsch.0.5.0] no changes from git+file:///home/ari/plotkicadsch#master
[plotkicadsch.0.5.1] no changes from git+file:///home/ari/plotkicadsch#master
Now run 'opam upgrade' to apply any package updates.
ari@omen:~/plotkicadsch$ opam upgrade
Everything as up-to-date as possible (run with --verbose to show unavailable upgrades).

The following packages are not being upgraded because the new versions conflict with other installed packages:
  - base.v0.12.2
  - bin_prot.v0.12.0
  - core_kernel.v0.12.2
  - fieldslib.v0.12.0
  - jane-street-headers.v0.12.0
  - parsexp.v0.12.0
  - patience_diff.v0.12.0
  - ppx_assert.v0.12.0
  - ppx_base.v0.12.0
  - ppx_bench.v0.12.0
  - ppx_bin_prot.v0.12.1
  - ppx_compare.v0.12.0
  - ppx_custom_printf.v0.12.1
  - ppx_enumerate.v0.12.0
  - ppx_expect.v0.12.0
  - ppx_fail.v0.12.0
  - ppx_fields_conv.v0.12.0
  - ppx_hash.v0.12.0
  - ppx_here.v0.12.0
  - ppx_inline_test.v0.12.0
  - ppx_jane.v0.12.0
  - ppx_js_style.v0.12.0
  - ppx_let.v0.12.0
  - ppx_optcomp.v0.12.0
  - ppx_optional.v0.12.0
  - ppx_pipebang.v0.12.0
  - ppx_sexp_conv.v0.12.0
  - ppx_sexp_message.v0.12.0
  - ppx_sexp_value.v0.12.0
  - ppx_typerep_conv.v0.12.0
  - ppx_variants_conv.v0.12.0
  - seq.base
  - sexplib.v0.12.0
  - sexplib0.v0.12.0
  - splittable_random.v0.12.0
  - stdio.v0.12.0
  - typerep.v0.12.0
  - variantslib.v0.12.0
However, you may "opam upgrade" these packages explicitly, which will ask permission to downgrade or uninstall the conflicting packages.
Nothing to do.
ari@omen:~/plotkicadsch$ opam install plotkicadsch

<><> Synchronising pinned packages ><><><><><><><><><><><><><><><><><><><><><><>
[plotkicadsch.0.5.1] no changes from git+file:///home/ari/plotkicadsch#master

Sorry, no solution found: there seems to be a problem with your request.

No solution found, exiting
ari@omen:~/plotkicadsch$ cd ~/global-sync/dish/pcb/
ari@omen:~/global-sync/dish/pcb$ plotkicadsch -l square-dish-cache.lib -f square-dish.sch 
plotkicadsch: command not found
```

jnavila, 2019-07-29:
Mmm... `patience_diff` should be installed in a version anterior to v0.12 . This last version is no longer compatible with Windows and I had to downgrade. I guess the other packages are directly version-dependent from `patience_diff`.

Try:

```shell
$ opam upgrade patience_diff
$ opam upgrade
```

arikrupnik, 2019-07-29:
I have no idea what I'm doing :=(

```
ari@omen:~/plotkicadsch$ opam upgrade patience_diff
Sorry, no solution found: there seems to be a problem with your request.
```

jnavila, 2019-07-29:
Sorry, I think you have to specify the version that is forced.

```
$ opam upgrade patience_diff.v0.11.0
```

jnavila, 2019-07-29:
The strange thing is that all these dependencies should be managed automatically.

arikrupnik, 2019-07-29:
I feel like such a complete noob.

```
ari@omen:~/plotkicadsch$ opam upgrade patience_diff.v0.11.0
Everything as up-to-date as possible (run with --verbose to show unavailable upgrades).

The following packages are not being upgraded because the new versions conflict with other installed packages:
  - patience_diff.v0.12.0
However, you may "opam upgrade" these packages explicitly, which will ask permission to downgrade or uninstall the
conflicting packages.
Nothing to do.
```

jnavila, 2019-07-29:
Don't worry, I feel like a noob too. Just throwing commands in the dark...

Last trial before asking you to create a new switch and restart from scratch.
```
$ opam install patience_diff.v0.11.0
```

arikrupnik, 2019-07-29:
OK, so that ran without errors, but I stilt can't get `plotkicadsch` to install:

````
ari@omen:~/plotkicadsch$ opam install plotkicadsch

<><> Synchronising pinned packages ><><><><><><><><><><><><><><><><><><><><><><>
[plotkicadsch.0.5.1] no changes from git+file:///home/ari/plotkicadsch#master

Sorry, no solution found: there seems to be a problem with your request.

No solution found, exiting
````

jnavila, 2019-07-31:
OK. If you are still interested, let's start afresh and create a new compiler switch:

``` shell
$ opam switch create plotkicad_git 4.07.0
$ eval `opam config env`
$ opam pin add kicadsch .
[kicadsch.0.5.1] synchronised from git+file:///home/jnavila/perso/plotkicadsch#master
kicadsch is now pinned to git+file:///home/jnavila/perso/plotkicadsch#master (version 0.5.1)

The following actions will be performed:
  ∗ install dune     1.11.0 [required by kicadsch]
  ∗ install kicadsch 0.5.1*
===== ∗ 2 =====
Do you want to continue? [Y/n] n
$ opam pin add plotkicadsch .
[plotkicadsch.0.5.1] synchronised from git+file:///home/jnavila/perso/plotkicadsch#master
plotkicadsch is now pinned to git+file:///home/jnavila/perso/plotkicadsch#master (version 0.5.1)
...
===== ∗ 110 =====
Do you want to continue? [Y/n] y
```

If you have followed the steps of the `README`, the old compiler switch is no longer needed:

```shell
$ opam switch remove 4.06.0
Switch 4.06.0 and all its packages will be wiped. Are you sure? [Y/n] y
```

jnavila, 2019-10-12:
This is fixed in v0.5.2

closed 2019-10-12 by jnavila

arikrupnik, 2019-10-16:
I'm sorry I was never able to test the fix, but I'm confident that this does resolve it. Thank you for being so responsive to bug reports and for your patience with my setup.