jnavila/plotkicadsch #28: git-imgdiff not executing

abdallahfalou, 2019-08-23:
Hi,

I've tried installing on Windows in Git Bash. Scripts are in Mingw64's bin dir. All seems fine with the internal diffing tool (and looks great, thanks!). 

If I run without the `-i` option, I get no errors but the output is simply the git-imgdiff file itself. That is, my viewer simply shows me the contents of git-imgdiff, which are:
`#!/bin/bash
PIPE=$(mktemp -u)
(! compare -metric RMSE $2 $1 png:${PIPE} 2> /dev/null) &&  (montage -geometry +4+4 $2 $PIPE $1 png:- | display -title "$1" -)
rm $PIPE`

Seems like git-imgdiff is being opened rather than executed by plotgitsch. Tried running `chmod +x` on git-imgdiff but that didn't solve it.

jnavila, 2019-08-24:
To be honest, I don't think a bash script can be run on Windows. I have no Windows system to test on, so giving advices is going to be difficult. Moreover, even with the git-imgdiff script running, you would still need the `imagemagick` package because the `compare`, `montage` and `display` utilities come from it.

In the same way as python does, what I would try first is to rename `git-imgdiff` to `git-imgdiff.sh` or similar: Windows does not understand executable bit, but uses the file extension to decide how a file should be opened.

abdallahfalou, 2019-08-24:
Thanks. I got this working. A bit hacky, but I'm sure it can be done better - I'm honestly not well-versed enough in bash and particularly within a Windows env. 

Here's what I did:

1. Download ImageMagick portable Windows binaries and add to PATH.
1. Rename `git-imgdiff` to `git-imgdiff.sh`. Associate *.sh extension with Git Bash. 
2. Create a Windows shortcut (_not_ an alias) `git-imgdiff` that points to `git-imgdiff.sh`. Both are in my PATH.
3. Modify `git-imgdiff.sh` to use `imdisplay` instead of `display`. This is the one tool not included in the Windows binaries of ImageMagick. `git-imgdiff.sh` now reads:
`#!/bin/bash
PIPE=$(mktemp -u)
(! compare -metric RMSE $2 $1 png:${PIPE} 2> /dev/null) &&  (montage -geometry +4+4 $2 $PIPE $1 png:- | imdisplay -title "$1" -)
rm $PIPE`

`imdisplay` is a very basic tool, so I'll look for an alternative that can be passed a pipe as you do above. Will report back if I find a better one. Also, I think I like the internal diff tool better with `-i`, but I faced a few issues there as well on Windows. Will submit a separate issue for that.

Thanks!

closed 2019-08-27 by jnavila