jnavila/plotkicadsch #31: Exception when calling plotgitsch --file

creat1c, 2020-04-08:
Hello,

I have set up everything so that I can successfully generate diff images of all schematics manually by entering the GIT project directory in a bash shell (mingw64, im on windows) and typing plotgitsch -ifirefox.

It somehow does not work when tying to limit to specific files, i.E. 

USER@XYZ MINGW64 /c/Dev/repo (develop)
$ plotgitsch -ifirefox --file=/c/Dev/repo/Hardware/schematic.sch a48b20d6f2782f5d2ad74804aab9289d15450a12
internal diff and show with firefox between Git rev a48b20d6f2782f5d2ad74804aab9289d15450a12 and file system .
Git Exception: path not found: /C:

Im not sure if I messed up with the paths, but I also tried windows paths and relative paths, and it always turns up a similar. Is it somehow related to window-isms?

jnavila, 2020-04-08:
You have to cd to the directory of the project and use relative path:
```
$ cd Hardware
$ plotgitsch -ifirefox --file=schematic.sch
```

creat1c, 2020-04-09:
Seems to work. Great!

What I was trying to do is call the script from the Tortoise GIT shell menu, This works in a way that one can assign a file extension with a custom command line, such as 

.sch: plotgitsch %yrev %brev %mine

Unfortunately its not as easy as directly calling the scripts, so I ended up with two script layers mangling some file name and getting the script to run in a bash shell. For anyone interested, I left the scripts here.

invoke-plotgitsch.bat:
```
@ECHO OFF

@SET FPATH=%~dp3
@SET FNAME=%~nx3
@SET FNAME_NOQ=%FNAME:"=%

@ECHO %FPATH%
@ECHO %FNAME%

@IF "%1" == ""0000000000000000000000000000000000000000"" @GOTO ONEREV
@GOTO TWOREV

:ONEREV
@CD %FPATH%
@C:\Dev\invoke-plotgitsch.sh "%FNAME_NOQ%" "%2" 
@GOTO END

:TWOREV
@CD %FPATH%
@C:\Dev\invoke-plotgitsch.sh "%FNAME_NOQ%" "%1" "%2" 
GOTO END

:END
PAUSE

```

invoke-plotgitsch.sh:
```
#!/bin/bash 
plotgitsch -ifirefox --file=$1 $2 $3 $4 $5
```

Then I used set up a call to the batch script in the TortoiseGIT settings (Diff Viewer -> Advanced -> Add, File extension=.sch , Command = C:\Dev\invoke-plotgitsch.bat %yrev %brev %mine)

closed 2020-04-09 by creat1c