jnavila/plotkicadsch #32: plotgitsch: unknown option `--lib'

Munken, 2020-04-24:
The version on opam does not allow once to specify --lib

Munken, 2020-04-24:
With a bit more detail:

```
 % plotgitsch -ifirefox -lVMEmaster-cache.lib prototype0 HEAD
plotgitsch: unknown option `-l'.
Usage: plotgitsch [OPTION]... [FROM_REF] [TO_REF]
Try `plotgitsch --help' for more information.
 % plotgitsch -ifirefox --lib=VMEmaster-cache.lib prototype0 HEAD
plotgitsch: unknown option `--lib'.
Usage: plotgitsch [OPTION]... [FROM_REF] [TO_REF]
Try `plotgitsch --help' for more information.
```

jnavila, 2020-04-24:
Which version are you using?

``` bash
[jnavila:~] 4s % plotgitsch --version
v0.5.2
[jnavila:~] 7s % plotgitsch --help | cat
PLOTGITSCH(1)                  Plotgitsch Manual                 PLOTGITSCH(1)

NAME
       plotgitsch - Show graphically the differences between two git revisions
       of a kicad schematic

SYNOPSIS
       plotgitsch [OPTION]... [FROM_REF] [TO_REF]

ARGUMENTS
       FROM_REF (absent=Git rev HEAD)
           reference from which the diff is performed. If it starts with
           'dir:' it's a file system dir.

       TO_REF (absent=file system .)
           target reference to diff with. If it starts with 'dir:' it's a file
           system dir.

OPTIONS
       -c old:new:foreground:background,
       --colors=old:new:foreground:background (absent=default colors or
       PLOTGITSCH_COLORS env)
           list of colon separated hex RGB codes for colors used for diffing
           e.g. the default colors are FF0000:00FF00:00000:FFFFFF

       -f FILENAME, --file=FILENAME
           diff only selected file FILENAME.

       --help[=FMT] (default=auto)
           Show this help in format FMT. The value FMT must be one of `auto',
           `pager', `groff' or `plain'. With `auto', the format is `pager` or
           `plain' whenever the TERM env var is `dumb' or undefined.

       -i [BROWSER], --internal[=BROWSER] (default=internal with viewer
       xdg-open) (absent=external or PLOTGITSCH_VIEWER env)
           use an internal diff algorithm and use the BROWSER to display the
           result.

       -k, --keep
           by default, the svg diff files are deleted after launching the
           viewer; this option lets the files in place after viewing them.

       -l LIB, --lib=LIB
           preload symbol library LIB in order to prepare the diff. This
           option can be used several times on command line.

       -t, --textdiff
           fall back to show a text diff if files are different but generate
           no visual diffs

       --version
           Show version information.

EXIT STATUS
       plotgitsch exits with the following status:

       0   on success.

       124 on command line parsing errors.

       125 on unexpected internal errors (bugs).

ENVIRONMENT
       These environment variables affect the execution of plotgitsch:

       PLOTGITSCH_COLORS
           Colors for plotting the diff

       PLOTGITSCH_VIEWER
           Default viewer for internal diff. Defining this env var forces
           internal diff.

BUGS
       Open issues to https://github.com/jnavila/plotkicadsch/issues

Plotgitsch v0.5.2                                                PLOTGITSCH(1)
```

Munken, 2020-05-27:
Sorry for the long wait. Got side tracked.

If I run
```
> opam switch 4.07.0
> eval `opam config env`
> opam update
> opam install plotkicadsch
> plotgitsch --version
v0.3.0
```
That seems a bit old?

Munken, 2020-05-27:
Hmm this seems to be a opam related issue:

```
 % opam show plotkicadsch                                        src/plotkicadsch (master) munk-zen2
             package: plotkicadsch
             version: 0.3.0
          repository: default
        upstream-url: https://github.com/jnavila/plotkicadsch/releases/download/v0.3.0/plotkicadsch-0.3.0.tbz
       upstream-kind: http
   upstream-checksum: 637f205fbe0c524716566a32b1756c64
            homepage: https://jnavila.github.io/plotkicadsch/
         bug-reports: https://github.com/jnavila/plotkicadsch/issues
            dev-repo: https://github.com/jnavila/plotkicadsch.git
              author: Jean-Noel Avila <jn.avila@free.fr>
             license: ISC
             depends: jbuilder & kicadsch = 0.3.0 & tyxml >= 4.0.0 & lwt & lwt_ppx & sha & git < 2.0.0 & git-unix & base64 & patience_diff >= v0.10.0 & core_kernel & cmdliner
   installed-version: 0.3.0 [4.07.0]
  available-versions: 0.1.4, 0.2.0, 0.3.0
         description: Schematic plotter
```

Munken, 2020-05-27:
Sorry for the spam. I resolved the issue.

Ubuntu 18.04 only provides opam v. 1. This is using a frozen version of the opam packages. 
One has to install a newer version of opam via a ppa as described here: https://askubuntu.com/questions/1080715/opam-2-0-on-ubuntu

A word of warning to other Ubuntu users could probably be nice.

closed 2020-10-20 by jnavila