jnavila/plotkicadsch #33: Find schematics in sub-folders

leardilap, 2020-05-03:
I have most of my kicad project organized in a similar structure where I have a folder where all schematic sheets are, with the exception of the top level:
```
├── images
├── mech
├── pdf
├── sch
│   ├── sub-sheet1.sch
│   ├── sub-sheet2.sch
├── project-cache.lib
├── project.kicad_pcb
├──project.pro
├──project.sch
```
whent I run `plotgitsch -i firefox` only differences in project.sch are shown.
in need to `cd sch/` and run `plotgitsch -i firefox -l ../project-cache.lib`

it is a minor annoyance, but I think it could be fix by finding all files ending in .sch including a search in folders.

I tried looking in to the code but I'm not familiar which language is this tool written on.
I guess it would be in line 48 of plotkicadsch/src/kicadDiff.ml

`let find_schematics () = F.list_files (ends_with ".sch")`

Many thanks for this tool, 
Luis

jnavila, 2020-05-03:
I did not know that you could organize your sch files in subdirs of the main sch and lib files. In such case, there is indeed a bug.

Thanks for reporting.

leardilap, 2020-05-07:
in Kicad when you create a sub-sheet.sch you need to place in the "file name" field: sch/sub-sheet.sch 
so it knows it is located under the folder sch/

commit 0b170a67 (2020-05-22): Update changelog for version v0.6.0

Fixes #33
closed 2020-05-22 by jnavila