jnavila/plotkicadsch #34: Help to install

leoheck, 2020-05-27:
It is always hard to install this. Can you help me?
Tried master version and the default version from the package manager

```
plotkicadsch/plotkicadsch on  master took 3s 
➜ opam install plotkicadsch      

<><> Synchronising pinned packages ><><><><><><><><><><><><><><><><><><><><><><>
[plotkicadsch.0.5.2] no changes from git+file:///home/lheck/Dropbox/kicad-plugins/plotkicadsch#master

Sorry, no solution found: there seems to be a problem with your request.

No solution found, exiting
```

leoheck, 2020-05-27:
I don't have this 4.06.0, should I create this?

[image]

leoheck, 2020-05-27:
Still no lucky 
```
➜ opam install plotkicadsch
[ERROR] Sorry, resolution of the request timed out.
        Try to specify a simpler request, use a different solver, or increase the
        allowed time by setting OPAMSOLVERTIMEOUT to a bigger value (currently,
        it is set to 60.0 seconds).
```

leoheck, 2020-05-27:
Any idea what I am doing wrong?

leoheck, 2020-05-27:
Is it possible to remove the $ in front of the commands? It will make the copy-paste processes so much easier?

leoheck, 2020-05-27:
Suppose I have a clean OS install, which is actually the truth right now. Does the steps cover the whole opam install?

I started with:
```
sudo ap install opam
```

Then I followed your steps..

leoheck, 2020-05-27:
Installing from the repo I got this

```
plotkicadsch on  master took 2m 17s 
➜ opam install plotkicadsch

<><> Synchronising pinned packages ><><><><><><><><><><><><><><><><><><><><><><>
[plotkicadsch.0.5.2] no changes from git+file:///home/lheck/Dropbox/kicad-plugins/plotkicadsch#master

Sorry, no solution found: there seems to be a problem with your request.

No solution found, exiting
```

leoheck, 2020-05-27:
Should I install dune, lwt_ppx and another package?
They are not installed by default and the install process does not install them..

leoheck, 2020-05-27:
Good, after some time, I made it. This was my process..

```
# Install 

opam init

opam install --switch 4.06.0 plotkicadsch
eval $(opam env)
opam switch 4.06.0
opam update

opam install -y --jobs=8 --deps-only .
opam install .
```

jnavila, 2020-05-27:
Thanks! Will update the guide with your procedure from scratch.

Anyway, the next version of plotkicadsch is on its way to opam.

leoheck, 2020-05-27:
> Anyway, the next version of plotkicadsch is on its way to opam.

Cool! And sorry for the number of unconnected messages. I was struggling to make it work very late at night.

leardilap, 2020-05-27:
just a comment: is opam 4.06.0 really the correct version? 
I see the travis file with OCAML_VERSION=4.07
in a clean Centos8 system I did this:

```
# install opam according to your distro, in my case (centos8) was
wget https://raw.githubusercontent.com/ocaml/opam/master/shell/install.sh
sudo sh install.sh

opam switch 4.07.0 # at this point it showed two errors \
[ERROR] Opam has not been initialised, please run 'opam init' \
[ERROR] No switch 4.07.0 is currently installed. Did you mean 'opam switch create 4.07.0'

# suggestion for instructions
opam init
opam switch create 4.07.0
eval `opam config env`
opam update
opam install plotkicadsch
```

leoheck, 2020-05-30:
Just to adding here a second try to install this using an almost clean Ubuntu docker 

```
# Schematic Diffs
run apt-get -y install --no-install-recommends m4
run apt-get -y install --no-install-recommends rsync
run apt-get -y install --no-install-recommends mercurial
run apt-get -y install --no-install-recommends darcs
run apt-get -y install --no-install-recommends opam
run git clone https://github.com/jnavila/plotkicadsch.git
run cd plotkicadsch
run opam init --disable-sandboxing
run opam switch create 4.06.0
run eval $(opam env)
run opam switch 4.06.0
run opam update
run opam install -y --jobs=8 --deps-only .
run opam install .
```

Maybe the last 2 lines can be reverted to the original `opam install plotkicadsch`

leoheck, 2020-05-30:
This is the issue I was having before. Trying to install it on another ubuntu machine.
[image]

leoheck, 2020-05-30:
Using this 4.07.0 works out of the box. @leardilap @jnavila 
> This is solves my problem too. Thanks,
https://github.com/jnavila/plotkicadsch/issues/34#issuecomment-634744276

Maybe this can accelerate the last step (for me it is installing 226 packages) :)
```
opam install -y --jobs=8 plotkicadsch
```

jnavila, 2020-05-30:
> Maybe this can accelerate the last step (for me it is installing 226 packages) :)
> 
> ```
> opam install -y --jobs=8 plotkicadsch
> ```

Building locally gets you the "development" version, whereas your stanza builds the version tagged in opam. @leardilap wants to have the "development" version which adds the ability to search for schematics in sub-directories of a project.

leoheck, 2020-05-30:
Oh, I see, I want to have this feature too. I was not able to install it in any way.
The trick was this `4.07.0` instead of this `4.06.0`. Maybe is good to update this in the README.md

closed 2020-10-20 by jnavila