jnavila/plotkicadsch #35: Changing colors

leoheck, 2020-05-30:
Hi @jnavila if you add this into the README where you show how to change colors would be nice.

```
-c REMOVED_RGB:ADDED_RGB:UNCHANGED_RGB:BACKGROUND_RGB
```

jnavila, 2020-05-30:
I described this feature at https://jnavila.github.io/plotkicadsch/plotgitsch_usersguide.html

Maybe it should do made more visible.

leoheck, 2020-05-30:
Yeah, maybe. 

IMHO the `README.md` is always the best place to add these initial info like the usage and the command line arguments available because most of all repositories have one, and it is fast to access info even after downloading the package.

But, hours later when I was able to install the repo successfully, I get a chance to find manual in the tool itself as it is expected (`plotgitsch -help`). Which was good.

closed 2020-10-20 by jnavila