jnavila/plotkicadsch #36: Flag to disable vizualization

leoheck, 2020-05-30:
Does it have any flag to generate images only? I am asking this to use with CI where I can't launch any image visualization tool

jnavila, 2020-05-30:
You can specify as a vizualization tool a no-op, such as `cat`.

leoheck, 2020-05-30:
Do you mean as in `-i touch` for instance?
That's a clever workaround.

leoheck, 2020-05-31:
When using the `-k` flag (and `-i touch`)  I saw in your documentation that a folder will be generated in the project folder. What is the name of this folder? During my tests nothings was generated. Does this is working for you? I am testing with plotgitsch `v0.6.0-1-ga97d931` now.

closed 2020-05-31 by leoheck

jnavila, 2020-05-31:
Just out of curiosity, what is the purpose of running `plotgitsch` in CI?

leoheck, 2020-05-31:
Good question. 

CI is just automation, for me. So, considering I am working with other people, with different skills, also with different Operating Systems. I would like to create a fast approach, for when a person is creating a PR, to be able to review changes made. 

For instance, it is similar to check the code formating with the CI. Users can do that on their PCs, before committing code, but the CI can help to double-check this process.

Similar to that, I can set everything easily on my computer. I can't do the same for others, especially because they don't ask for help.

So, my idea is, when changes are being pushed, the CI makes all of these and delivers the results back to the GitHub so reviewers can be faster.

Right now, it looks promising, I am doing this for the pcb layouts using Kicad-Diff. I also added a PR there to add an initial command line for their scripts. 

I was not able to make it work with the plotgitsch yet. I can generate diffs on my computer but not in the CI. I am using the flat `-k` but nothing is being generated

I would like to suggest that this flag could be improved like with `-k[=folder_path]`
So, you can use only `-k` to generate images inside the project, or `-k=./folder` to generate it in a different path. The `-k=/tmp` should work too. So, it the folder exists it generates files there. If not, it creates the folder first.