jnavila/plotkicadsch #37: Feature idea - HTML output

leoheck, 2020-05-31:
I would like to suggest to improve the output of plotgitsch as the [Kicad-Diff](https://github.com/Gasman2014/KiCad-Diff) does. If you didn't use it, there are some images in the end of their README. To be fair, these images are not good to show how their output is good.

So, It generates an html to display the generated diffs in an organized way

For plotgitsch, the user can enable this with the command line if you think it is not a good idea as default.

So, this html, could organize sch sheets and also display the text diff together.

Kicad-Diff, displays the main diff image on top, then bottom-right and the bottom-left the original images used in the diff process ([this](https://github.com/Gasman2014/KiCad-Diff/raw/master/Documents/composite.png)). There, this is done for each layer of the PCB. But for the schematic, this can be generated for each page.

jnavila, 2020-05-31:
I have had this project for a year. Ocaml provides libraries to develop lightweight webservers and javascript webpages. But I haven't achieved anything so far.

Immortalin, 2020-05-31:
https://github.com/openscopeproject/InteractiveHtmlBom

^^this might be helpful
And you can compile it to run in the browser too:
https://github.com/ocsigen/js_of_ocaml

Immortalin, 2020-05-31:
https://github.com/openscopeproject/InteractiveHtmlBom

^^ this might be relevant
You can probably compile plotkicadsch to run in browser too with https://github.com/jchavarri/jsoo-react

leoheck, 2020-05-31:
A webserver is nice, but it is not so important. Being able to have a folder with an index.html as a starting point to organize and orient the review process would be a nice feature to have. But of course, after that a webserver can be nice, to share the view with the colleagues in the network to improve the review.

Considering the CI, I am trying to find simples a way to access these HTML files as a link right from the GitHub page, so users won't need to download the package and unzip them to be able to review this.

Kicad users like us could start some ideas to have actions for Kicad-related tasks.

leoheck, 2020-06-01:
@Immortalin this demo is really good! That is nice.
https://openscopeproject.org/InteractiveHtmlBomDemo/

leoheck, 2022-02-17:
I am closing this since it was a pretty old idea that was not pushed forward.

closed 2022-02-17 by leoheck