jnavila/plotkicadsch #38: I can't install it again.

leoheck, 2020-06-04:
Why is it so hard to install this tool?

Now, I can install neither the master version or the stable version.

I am installing it now inside a container

```
  - install patience_diff           v0.13.0 [required by plotkicadsch]
  - install plotkicadsch            0.6.1
===== 117 to install =====
Do you want to continue? [Y/n] n
The command '/bin/sh -c opam install plotkicadsch' returned a non-zero code: 10
make: *** [Makefile:12: build_release] Error 10
```

With this instructions
```
run opam init -c 4.09.0 --disable-sandboxing -a
run eval $(opam config env)
run opam update
run opam install plotkicadsch
```

jnavila, 2020-06-05:
I'm not sure at all that the environment variables set by `run eval $(opam config env)` are preserved in the following shell invocations. Can you try:

```
run eval $(opam config env) && opam update && opam install plotkicadsch
```

leoheck, 2020-06-05:
True, good point but still the same.

[image]

leoheck, 2020-06-05:
Sure, you are right, it is good to have is on a single line like this.

```
run opam init -c 4.09.0 --disable-sandboxing -a && \
    opam switch 4.09.0 && \
    eval $(opam config env) && \
    opam update && \
    opam install -y plotkicadsch
```

My biggest issue was solved by the `-y` in the last line otherwise the command fails without an answer.

Your instructions are working just fine. I also upgraded it on my computer. Thank you.

closed 2020-06-05 by leoheck

jnavila, 2020-06-06:
By the way, could you share your Dockerfile? That would be quite a good executable specification to help people build their own binaries.

leoheck, 2020-06-06:
Sure! I am trying to understand what is needed to allow building from the master version. Then I will add here a link to my Dockerfile.

Right now I am fighting with these

```
-> installed uri-sexp.3.1.0
[ERROR] The compilation of bos failed at "/root/.opam/4.09.0/bin/ocaml pkg/pkg.ml build --dev-pkg false".
[ERROR] The compilation of conduit failed at "/root/.opam/4.09.0/bin/dune build -p conduit -j 11".
[ERROR] The compilation of ppx_base failed at "/root/.opam/4.09.0/bin/dune build -p ppx_base -j 11".
[ERROR] The compilation of cohttp failed at "/root/.opam/4.09.0/bin/dune build -p cohttp -j 11".
```

leoheck, 2020-06-06:
[Here](https://github.com/leoheck/kicad_auto/blob/master/Dockerfile), it is. It is my first time creating a docker image, but it is working.

This Dockerfile is pretty much not organized yet since I am just evaluating the idea to use CI for hardware.

Considering `plotkicadsh` installation I am using the following commands now. The most important thing is not to forget to answer `-yes` whenever is necessary otherwise the build will fail since it does not have a way to accept stdin. Also, some packages are required to use opam (they are at the beginning of the Dockerfile) 

```Dockerfile
#=======================================
# jnavila/PlotKicadSch
#=======================================

# STABLE VERSION
# run opam init -c 4.09.0 --disable-sandboxing -a && \
# 	opam switch 4.09.0 && \
# 	eval $(opam config env) && \
# 	opam update && \
# 	opam install -y plotkicadsch

# GITHUB VERSION
run git clone https://github.com/jnavila/plotkicadsch.git && \
	cd plotkicadsch && \
	opam init -c 4.09.0 --disable-sandboxing -a && \
	opam switch 4.09.0 && \
	eval $(opam config env) && \
	opam pin add -y kicadsch . && \
	opam pin add -y plotkicadsch . && \
	opam update -y && \
	opam install -y plotkicadsch

run ln -sf /root/.opam/4.09.0/bin/plotgitsch /usr/local/bin/plotgitsch
```

jnavila, 2020-06-07:
Thanks for the input. All this electronics continuous integration stuff was a good discovery.

If I could give a piece of advice : you had better build plotgitsch in a dedicated image and only copy the output to the Dockerfile you intend to use for your CI. This is using [multi-stage builds](https://www.ardanlabs.com/blog/2020/02/docker-images-part1-reducing-image-size.html) .That would make your final docker image lighter.