jnavila/plotkicadsch #39: Flag to keep files is not working for me

leoheck, 2020-06-06:
I mean, it does not keep generated files inside the project folder. 
It is keeping files on /tmp 
Do we have any option to set the output path?

Right now I am seeing this `/tmp/diff_boardee37d8.svg` where the `ee37d8` part is a random number.

So, I would suggest this approach

```-k [otuptu_folder]```

jnavila, 2020-06-06:
Indeed there's a bug in the keep feature. Let me fix that first.

closed 2020-06-06 by jnavila

leoheck, 2020-06-06:
Cool! Thank you for looking into this. When you fix this, it will go to the stable opam version? or I have to install it from the branch? I am asking this because I am interested in using it in my container. So I will have to change the way it is installed. By now I am installing the server's version.

jnavila, 2020-06-06:
For now, I won't make a new tag. It impacts only MacOS and Linux users who can easily rebuild.

leoheck, 2020-06-06:
Which is my case (linux), so, is it on the master, right?

jnavila, 2020-06-07:
yes, it is. The way you are pinning the local copy should bring you the latest master version.

commit 9803f0ec (2020-06-06): Fix bug with keep option.

A bug was introduced with recursive search where the tmp files would
be moved to the root of filesystem. But anyway, using rename does not
work across mount points so this option is still brittle.

We open the diff files with different filenames depending on whether
the file needs to be kept or not.

Fixes #39