jnavila/plotkicadsch #40: Support new .kicad_sch format

ghent360, 2021-01-24:
Hi,

Looks like eschema is switching to a new file format. Any chance you can add support for it. I would help, but ocaml is way too foreign for me.

jnavila, 2021-01-25:
I've started having a look at supporting the new format. This will take some time.

jnavila, 2021-01-25:
By the way, the new format comes with the new version of Kicad which will be scriptable, make basically plotkicadsch useless.

leoheck, 2021-03-03:
What do you mean by scriptable? Do you think it is possible to generate a schematic image with python as we print layouts?
If this is is true, I think we should do something to add this on Kidiff too. Then the tool will be complete.

But I think this will take a long time to be implemented/available. v6 will have pretty limited support for eeschema in python and the whole library was pushed to v7. And you know how long these versions of Kicad take to be released. We can add 1,5 years at least.

I think the suggestion here is valid since Kicad 5.1.9 is already generating kicad_sch files.

How hard would be to implements this @jnavila? I am not sure how this is done here, I am assuming it is not hard. But I can be wrong.

jnavila, 2021-03-03:
> But I think this will take a long time to be implemented/available. v6 will have pretty limited support for eeschema in python and the whole library was pushed to v7. And you know how long these versions of Kicad take to be released. We can add 1,5 years at least.

Yes, I may have been too optimistic...

For the new kicad_sch file format, I'm working on it, but the lack of clear specification of the format is a big road block. The specification document no longer reflects the actual format. The format is full of unused features and I don't have a lot of spare time to sort out what's actually useful so it's going to take some time. But I'm not overly pessimistic, because from what I see, only the fields needed to draw the same primitives as the old format are actually used.

leoheck, 2021-03-03:
Good to know. 

I came here to say that if we could find something that converts this new format to the old format we could use it as the initial way to make it work for both formats. But I am not sure this exists.

Also, I am not sure, but I think Kicad 5.1.9 is generating both formats at once. 

I have a small board that has these 2 files, I am linking here for reference in case you need to check something inside. I believe both are in sync with each other.
https://github.com/leoheck/assoc-board/blob/main/board.sch
https://github.com/leoheck/assoc-board/blob/main/board.kicad_sch

leoheck, 2021-06-11:
> The format is full of unused features and I don't have a lot of spare time to sort out what's actually useful so it's going to take some time.

I could try to find projects with the new format then we could use them as starting point to have this new feature implemented? I am eager to move my projects to Kicad 6 (or 5.99) since this is a completely new world full of good features.

What do you think? 
Do you have a branch with the current work?

This new branch could add a flag to enable this new format/parser while it is a work in progress... then users could bring new issues as they will be appearing, instead of providing a fully working parser all at once.

leoheck, 2021-06-11:
Some examples here https://gitlab.com/search?search=kicad_sch&group_id=6593436&project_id=15502567&scope=&search_code=true&snippets=false&repository_ref=master&nav_source=navbar

jnavila, 2021-06-15:
Here it is: https://github.com/jnavila/plotkicadsch/tree/new_kicad

leoheck, 2021-06-15:
Nice, thank you. Found this now, in case someone is interested: [eeschema_schematic_file_format_EN](https://docs.google.com/document/d/1AiPZvWT_QBPyVQSyn3ISl6jVntjrrtATEYxzBFJM1HU/edit).

leoheck, 2021-06-18:
@jnavila I tested this when you shared this 3 days ago and it I thought I saw it working but now it is not creating anything on a simple board that I am using to test. Does this feature need to be activated with a flag?

jnavila, 2021-06-18:
For now, I haven't finished yet the parser, so, yes... It does not produce anything.

jnavila, 2021-06-18:
Please note also that the new format has more than just the file format. There's no longer need for a cache library, because the cache is embedded in each file. OTOH, the annotation of parts is no longer contained in the schematic file but is available in the profile (if I understood correctly). So that is a completely different processing of files.

leoheck, 2021-06-18:
Indeed, this new file ends with the cache-lib which is better.
For the annotation, I am not sure this happens.
I created one project here with a single component. One resistor `R1`.

I could only see references to `R1` in `.kicad_pcb and .kicad_sch` as you can see

```bash
kicad-v6-project on  master took 4s 
➜ grep R1 -R *                 
board.kicad_pcb:  (net 1 "unconnected-(R1-Pad1)")
board.kicad_pcb:  (net 2 "unconnected-(R1-Pad2)")
board.kicad_pcb:    (fp_text reference "R1" (at 0 -1.27) (layer "F.SilkS")
board.kicad_pcb:      (net 1 "unconnected-(R1-Pad1)") (pintype "passive") (tstamp 00d8cf93-ce0c-4c2b-9990-7e3df08bbc91))
board.kicad_pcb:      (net 2 "unconnected-(R1-Pad2)") (pintype "passive") (tstamp a1ee7c76-d7bd-48a2-8ac7-c25f87091595))
board.kicad_sch:    (property "Reference" "R1" (id 0) (at 157.9881 97.6641 0)
board.kicad_sch:      (reference "R1") (unit 1) (value "100") (footprint "Resistor_SMD:R_0201_0603Metric")

kicad-v6-project on  master 
➜ ls
board-backups  board.kicad_pcb  board.kicad_prl  board.kicad_pro  board.kicad_sch  fp-info-cache 
 ```

jnavila, 2021-06-19:
I recall badly. Annotations are in the root schematic. Here is the link: https://forum.kicad.info/t/kicad-nightly-v5-99-new-schematic-and-symbol-library-file-formats-are-now-the-default/22655 . See the 5th point

Gasman2014, 2021-06-19:
I did ask about eeschema scripting a while back on the forum. I was told that this is still planned for version 6 according to @craftyjon.

craftyjon, 2021-06-19:
It is still planned for V6 although I don't have any updates on timelines yet.  I would assume that it may take some iteration to get everything into the new API that is needed, so I don't think it will instantly make tools that parse the schematic files directly obsolete.

> Also, I am not sure, but I think Kicad 5.1.9 is generating both formats at once.

This is not true, the new file format is only in 5.99 (future 6.0).  If you see both files, it's likely because you have 5.99 installed next to 5.1 and you opened an existing project and saved: this will result in 5.99 saving the `kicad_sch` files next to the old `sch` files.

leoheck, 2021-06-20:
Yeah, this seems correct. I might have ran both stable and nightly version of the Kicad in the same project and then I had the wrong conclusion.

leoheck, 2021-11-18:
A note from a Kicad dev
https://forum.kicad.info/t/attention-python-devs-6-0-api-and-plugin-repository-info/32035

It looks like we are not going to have a way to export .kicad_sch to .svg soon on Kicad.
Having plotkicadsch updated for Kicad 6.0 would be awesome! 

@jnavila thoughts?

leoheck, 2021-11-18:
I am linking here the work in progress of this new `sexpr-schematic`
https://dev-docs.kicad.org/en/file-formats/sexpr-schematic/

leoheck, 2021-11-18:
Nice, if there is anything that I can do to help you. maybe testing current work.. let me know.

leoheck, 2021-12-28:
An interesting observation here is that if plotgitsch is going to support Kicad-v6, then it would be good that the support of Kicad-v5 still remains in the same tool. Since we are going to see some migrations from 5 to 6, and it will be possible to compare different versions of the schematics during this migration to the new format.

jnavila, 2021-12-29:
I was planning to keep the V5 in the tool. But, allowing a comparison between V5 and V6 is another story. The filenames are changed so the filename matching needs to be generalized. I'm still struggling with the parser...

leoheck, 2021-12-29:
Oh. That's it fine. The parser may be pretty hard to make.