jnavila/plotkicadsch #44: Keep original files.

leoheck, 2021-03-14:
The option to keep svg keeps only the diff. Is it possible to keep the 2 original versions too?

leoheck, 2021-03-14:
Can you guide me on what I should do to be able to do that? This language is so complex haha.

leoheck, 2021-03-14:
I would like to test how this can work with Kidiff, but I need to get original images instead of the final diff.

jnavila, 2021-03-14:
If you use the internal diff, then none of the original version are generated, because this isn't the internal format used to compute the difference. If you use the external diff (which would be the good way), the versions are generated on disk, ready to be diffed, so it is just up to you to modify the diffing script in order to put both created files into their respective directories. You don't need to fiddle with the source of plotgitsch in the last case.

leoheck, 2021-03-15:
I am not aware of those internal/external diffs.
I just want to have the svg from both commits and also the final svg with the diff as the outputs of the tool.
For instance, the -k flag could save these extra 2 files for the user. what do you think?
If it is not needed to change the tool, Can you show me an example of the command line to generate these extra svgs?

leoheck, 2021-03-19:
So, I never knew I could omit `-i` flag. Alright, doing this it is opening with the image with the ImageMagic tool it is creating these 2 images `from_board.svg` and `to_board.svg`. But ImageMagickc remains open. How can I close ImageMagick automatically or avoid it from opening? Any idea?

leoheck, 2021-03-19:
For instance, this is the output window that remains open after the operation.
IT is called  from_board.svg, but it has the 3 things. from, to, and dff.
[image]

jnavila, 2021-03-21:
Hello,

Sorry to be late and not responsive. You are free to define the diffing program as you want. For instance, instead of generating the diff and displaying it, you can simply copy the svg files to another location. 

Combining a call with the `-i` option in order to get a diff image and one without the `-i` option with a selected "comparison" program would get you the two versions and the diff.

leoheck, 2021-03-21:
Sorry, but I believe this is not right. 

The `-i` flag is not passing a diff tool but as a visualization tool.

For instance, your examples have this `-ifirefox` then it opens the generated file which is `diff_board.svg` only.

I can use the `-iinkscape` for instance to open it directly on Inkcaspe.
I can also use `-iecho` then it will display the parameters this `-i` flag is going to take from `plotgitsch` which is `diff_board.svg` only.

If this `-i` was considering a diff tool instead of a visualization tool it would pass 2 files, (`to_board.svg` and `from_board.svg`). Unless this is a bug which makes sense. 

For instance, for the sake of comparison, I would like to use the textual `diff` as the diff tool. Then I would use like `-idiff` to have a textual diff from these 2 image files. But this wont work since it is taking only one argument.

[image]

leoheck, 2021-03-21:
Also, I would like to keep the original svgs of each board even if we don't have any diff. We could have a flag for that.

leoheck, 2021-03-21:
I was able to make a version that generates these images even if we don't see any diff.
I was not able to fix this -i flag. Opam is too complex for me. I was also able to change `git-imagediff` to save/keeep 3 svgs at the end. But having this by default on this repo would be nice. At least fixing this `-i` thing that is not a diff tool but a visualization tool. Or by adding another flag to be able to set the diff tool, then I would be able to make a script that saves these images instead of making the diff.

jnavila, 2021-03-21:
Sorry, I wasn't clear enough. About the diffing program, I was the referring to the case you don't use the -i option. Then the `git-imgdiff` need not be the one that is provided by the project. If you override the "$PATH" environment variable and provide a `git-imgdiff` of your own with the behaviour that you desire, then you can have access to svg files of both revisions.

jnavila, 2021-03-21:
> Or by adding another flag to be able to set the diff tool, then I would be able to make a script that saves these images instead of making the diff.

That's a good option. I'll have a look at doing it (don't expect it soon though).

leoheck, 2021-03-21:
> Sorry, I wasn't clear enough. About the diffing program, I was the referring to the case you don't use the -i option. Then the git-imgdiff need not be the one that is provided by the project. If you [...]

Oh, ok. This makes sense. Thanks for explaining.  
Alright, I am creating my version of `git-diffimg` and pre-pending this on PATH.
This will work for me. Thank you one more time.

closed 2021-03-21 by leoheck