jnavila/plotkicadsch #46: git is changing to the master branch

leoheck, 2021-03-26:
Hey @jnavila I know you are super busy, but I just want to have your opinion on this since you understands Opam and your project and I don't.

I changed `gitplotshich` to generate and keep the original .svg files even if commits don't have any difference. This is just one or two lines of code.

The issue I am facing is that the git repository is changing the branch automatically for some reason. It is always moving to the `master' branch. In one project it dosn't have `master` it onoly has `main` and the other project I am working on a different branch.

So, since I was not able to track where the issue is. Could you check this code, to see if I am doing something that is not right? 
My idea was just to remove that if on line 141.

[image]

The comparison is here if you need more space to check.

leoheck, 2021-03-26:
Or maybe this code has nothing to with the thing I want to do. hehe

jnavila, 2021-03-26:
It has nothing to do with what you wanted to do :laughing: . This part of the code is for displaying the textual diff in case the visual diff is null (both files are identical).  The test is checking that the user required the textual diff via the command line option `-t`.

As a matter of fact, it seems that my last change (after I discovered I could miss some diffs if the sch files are identical but the libs have changed) makes `plotgitsch` draw the diffs all the time.

As for the `master` branch issue, the ocaml-git library is only an explorer of the git repository files. To my knowledge, it does not change anything to your working directory. `plotgitsch` retrieves files content from the database, and only uses git to parse revisions. I cannot be 100% sure, but the changes to the working tree are most probably not done here.

jnavila, 2021-03-26:
I spoke too quick. `plotgitsch` indeed changes to `master`... How come ?

jnavila, 2021-03-26:
I opened a ticket: mirage/ocaml-git#489

leoheck, 2021-03-26:
This is nice. Thank you.

jnavila, 2021-03-26:
The patch mirage/ocaml-git#490 solves the issue.

leoheck, 2021-03-27:
Interesting how much code was added to fix this. The workaround I was doing was to save `.git/HEAD` before executing plotgitsch and restoring it right after.

leoheck, 2021-04-01:
This https://github.com/mirage/ocaml-git/pull/490 merged. Should I do something here? Update any version? How does this work to propagate these changes?

jnavila, 2021-04-01:
It's  been merged but not tagged yet. If you want to test with this feature, clone the repo and tag `carton`, `carton-git`, `carton-lwt`, `git` and `git-unix` in this repo, refresh and update your opam packages and reinstall `plotgitsch`.

leoheck, 2021-04-02:
Yeah, I want to do that. Thank you for these instructions. I will let this issue opened so you can track this but feel free to close it if you want.

jnavila, 2021-04-25:
Good news! Upstream is available in opam: https://github.com/ocaml/opam-repository/pull/18558

closed 2021-04-25 by jnavila

leoheck, 2021-05-06:
Hey @jnavila how should I use this? Should I update something here? I am not sure how this fix will end in my machine.

leoheck, 2021-05-06:
Oh, I think I got it. https://github.com/jnavila/plotkicadsch/pull/48