jnavila/plotkicadsch #50: Nested projects

leoheck, 2021-05-17:
Hey, @jnavila it is me again.

Quick question. I have noticed something strange when having nested projects. 
First of all, I am using a custom `git-diff` that keeps all svg files even if they don't have any difference.
I have some duplicated files as you can see.. just to track my changes.

```bash
✦ ➜ cat ../plotkicadsch_lheck/plotkicadsch/src/git-imgdiff

#!/bin/bash

# New files (rename for not overriding multiple schematic pages)
# Remove 6 unknown digits/hashs from the filename
new1=$(echo $1 | sed "s|/tmp/from_\(.*\)\([0-9a-f]\{6\}\).svg|sch1-\1.svg|g")
new2=$(echo $2 | sed   "s|/tmp/to_\(.*\)\([0-9a-f]\{6\}\).svg|sch2-\1.svg|g")

hash1=$(echo $1 | sed "s|/tmp/from_.*\([0-9a-f]\{6\}\).svg|\1|g")
hash2=$(echo $2 | sed   "s|/tmp/to_.*\([0-9a-f]\{6\}\).svg|\1|g")

# Almost original filename, remove /tmp string and keep the random number
orig1=$(echo $1 | sed "s|/tmp/from_\(.*\)\([0-9a-f]\{6\}\).svg|sch1-\1-\2.svg|g")
orig2=$(echo $2 | sed   "s|/tmp/to_\(.*\)\([0-9a-f]\{6\}\).svg|sch2-\1-\2.svg|g")

echo "Schematic Page: $orig1 -> $new1 (${hash1})"
echo "Schematic Page: $orig2 -> $new2 (${hash2})"

cp -f "$1" "$new1"
cp -f "$2" "$new2"

# Keep originals
mv -f "$1" "$orig1"
mv -f "$2" "$orig2"

echo
```

I have multiples projects inside the repository. Each project has multiples schematic sheets.

For instance, I have something like this.

```
repo
  - proj1
     - board.sch
     - page1.sch
  -  proj2
     -  board.sch
     -  page2.sch
  -  proj3
     -  board.sch
     - page2.sch
```

So, everything is working pretty fine since most of the time schematic sheets do not have the same name, except for the main schematic sheet that is called `board.sch` on all of the projects.

This is the current output I have with my `git-diff` script
```bash
plotgitsch -l main-board/board-cache.lib -c FF0000:008FFF:FFFFFF:222222 a4da48d 56b5c9a #&> /dev/null
Use compare (ImageMagick) between bitmaps between Git rev a4da48d and Git rev 56b5c9a
Schematic Page: sch1-board-1ae151.svg -> sch1-board.svg (1ae151)
Schematic Page: sch2-board-30817f.svg -> sch2-board.svg (30817f)
Schematic Page: sch1-board-0b26c3.svg -> sch1-board.svg (0b26c3)
Schematic Page: sch2-board-5b2755.svg -> sch2-board.svg (5b2755)
Schematic Page: sch1-board-c002f3.svg -> sch1-board.svg (c002f3)
Schematic Page: sch2-board-5d8be5.svg -> sch2-board.svg (5d8be5)

Schematic Page: sch1-board-59c898.svg -> sch1-board.svg (59c898)
Schematic Page: sch2-board-3446cb.svg -> sch2-board.svg (3446cb)

Schematic Page: sch1-power_supply-7f4b49.svg -> sch1-power_supply.svg (7f4b49)
Schematic Page: sch2-power_supply-15b200.svg -> sch2-power_supply.svg (15b200)
Schematic Page: sch1-peripherals-ec9b8a.svg -> sch1-peripherals.svg (ec9b8a)
Schematic Page: sch2-peripherals-6a7f5f.svg -> sch2-peripherals.svg (6a7f5f)
Schematic Page: sch1-sensor-4437c2.svg -> sch1-sensor.svg (4437c2)
Schematic Page: sch2-sensor-c52aac.svg -> sch2-sensor.svg (c52aac)

Schematic Page: sch1-microcontroller-719496.svg -> sch1-microcontroller.svg (719496)
Schematic Page: sch2-microcontroller-6b2627.svg -> sch2-microcontroller.svg (6b2627)
```

You can easily see that are many duplicated files (`sch1-board-*` ,`sch2-board-*`). This happens because `plotgitsh` is finding these guys on all of my projects.

So, my question is, does `plotgitsch` have a way to limit its scope to the current folder and not to its parent?

jnavila, 2021-05-17:
I haven't tested, but did you try to change current directory at the project level?

leoheck, 2021-05-17:
I haven't tested it too. I use to run things from the repo's root.
But checking here and it works, thanks for the suggestion.

What do you think about adding a flag to pass the project inside of the repo? Is it possible? 
I would like to propose improvements here with a PR but I cannot read OCaml yet.

jnavila, 2021-05-18:
This is also doable. I'll look into providing it...

Thank you for keeping me busy :wink:

leoheck, 2021-05-18:
:) It is my pleasure haha.

leoheck, 2021-05-21:
Testing this new relative path here. 
I found a small issue, maybe related to the symlinks

This is my folder structure
```bash
➜ tree -L 1
.
├── kicad-library (my custom parts)
├── my-board
│   └── library (symlink to ../kicad-library)
├── my-board2
│   └── library (symlink to ../kicad-library)
└── README.md
```

```bash
➜ plotgitsch -r ./my-board -l my-board/board-cache.lib
plotgitsch: internal error, uncaught exception:
            (Sys_error
              "././my-board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library: Too many levels of symbolic links")
            Raised by primitive operation at file "plotkicadsch/src/trueFs.ml", line 40, characters 21-39
            Called from file "plotkicadsch/src/trueFs.ml", line 50, characters 6-19
            Called from file "plotkicadsch/src/kicadDiff.ml", line 122, characters 20-42
            Called from file "cmdliner_term.ml", line 25, characters 19-24
            Called from file "cmdliner.ml", line 117, characters 32-39
```

leoheck, 2021-05-21:
This is a second repo, with the files all in the root path.

But I am tyring the relative path anyways. I am not sure if I am doing it right but I think it should work. 
➜ plotgitsch -m -r ./ -l board-cache.lib HEAD~1 HEAD~3
Use compare (ImageMagick) between bitmaps between Git rev HEAD~1 and Git rev HEAD~3
Exception GitFs.PathNotFound(0)

This is my current workaround to go inside the folder, it works. Because ti is going inside `./` in other words, doing nothing.
```
assoc-board on main [?] 
➜ plotgitsch-dir -d . -m -l ./board-cache.lib fcd81ef 0f299ec
plotgitsch -l ./board-cache.lib -m fcd81ef 0f299ec
Use compare (ImageMagick) between bitmaps between Git rev fcd81ef and Git rev 0f299ec
Exporting...
sch1-board.svg (66a393)
sch2-board.svg (be7177)
```

This is using the new `-r` flag, but setting it to the current directory.
Perhaps it was not designed wit this in minde this since it is not needed in this case.
But I believe it should work with "./" too.
```
assoc-board on main [?] 
➜ plotgitsch -r . -m -l ./board-cache.lib fcd81ef 0f299ec 
Use compare (ImageMagick) between bitmaps between Git rev fcd81ef and Git rev 0f299ec
Exception GitFs.PathNotFound(0)
```

leoheck, 2021-05-21:
This test is using the nested project from my first reply.

```
# Here I just replace the symlinked library with the real one
➜ mv ./my-board/library ./my-board/library.old.symlink
➜ cp -r ./kicad-library ./my-board/library

# Testing again. It doesn't find a path, not sure which one. Maybe because the real library was not found in the previows versions of the git history.
➜ plotgitsch -r ./my-board -l my-board/board-cache.lib HEAD HEAD~3
Use compare (ImageMagick) between bitmaps between Git rev HEAD~2 and Git rev HEAD~3
Exception GitFs.PathNotFound(0)
```

jnavila, 2021-05-21:
> plotgitsch: internal error, uncaught exception:
>             (Sys_error [...]

This is a sign that your repository is recursing on itself. Unfortunately, my code is blind to this kind or recursion and chokes. If you try `ls -lR`, do you experience the same kind of issue?

jnavila, 2021-05-21:
> This is a second repo, with the files all in the root path.
> 
> But I am trying the relative path anyways. I am not sure if I am doing it right but I think it should work.
> 
 Ah, you got me. The code does not handle this form of relative path (without real subdir, and with ./ in it), only plain 'subdir1/subdir2'.

jnavila, 2021-05-21:
> This test is using the nested project from my first reply.
>
> ```
> # Here I just replace the symlinked library with the real one
> ➜ mv ./my-board/library ./my-board/library.old.symlink
> ➜ cp -r [...]

try `plotgitsch -r my-board -l my-board/board-cache.lib HEAD HEAD~3`

jnavila, 2021-05-21:
I have to clean up the relative path a bit before consumption.

leoheck, 2021-05-21:
The library inside each kicad project is a symlink. It refers to a folder inside the root (one level up)
```
library -> ../kicad-library
```

The recursion is being generated inside plotgitsh. Maybe something inside does not handle well symlinks.

Inside the project folder
[image]

Inside the repo root
[image]

leoheck, 2021-05-21:
Oh, sorry for the mistake. 2 files were not properly update during the `git pull`. 
It works.

```
plotgitsch -m -r my-board -l my-board/board-cache.lib HEAD HEAD~3
use compare (ImageMagick) between bitmaps between Git rev HEAD and Git rev HEAD~3
Exporting...
sch1-board.svg (0dce0a)
sch2-board.svg (42b010)
```

jnavila, 2021-05-22:
Can I close the issue then?

leoheck, 2021-05-22:
Sure. I kept open since you said something about managing the `-r .` case. I will let you decide then, cool?

closed 2021-06-07 by jnavila

Javidgg, 2022-10-22:
> I have to clean up the relative path a bit before consumption.

Javidgg, 2022-10-22:
> The library inside each kicad project is a symlink. It refers to a folder inside the root (one level up)
> ```
> library -> ../kicad-library
> ```
>
> The recursion is being generated inside plotgitsh. Maybe [...]

leoheck, 2022-10-22:
What is this?

jnavila, 2022-10-23:
Probably a bot trying to go through the anti-spam filters.

commit 25c14ffb (2021-05-21): plotgitsch: introduce --relative option

fix #50
commit 69a2cd49 (2021-06-07): Allow relative paths to start with "./"

fixes #50