jnavila/plotkicadsch #53: Commit info

leoheck, 2021-05-20:
I am playing with the new changes you made.

I am seeing this. 
One of the commits shows this "Git rev HEAD"
[image]

The second one does not have any version info. Is this intentional?
Maybe it is good to have it in the second file too.
[image]

However this one has this `file system .` which may not be the thing you wanted, I am not sure.

leoheck, 2021-05-20:
Ah, I see. It happened because I was running `plotgitsh` with no parameters. Maybe this was a corner case.

Running it passing both versions work fine, as I could see.

```
 plotgitsch HEAD~5 HEAD~3
```

[image]
[image]

I really like the way they are separated so they won't be always overlapped in the diff image.

leoheck, 2021-05-20:
Ah, it comes from here. Got it. Maybe you want to polish this behavior, or maybe not.

Alternatives if you wish. 
`Git rev [filename].sch`
`Git rev unstaged`

[image]

jnavila, 2021-05-20:
I don't understand what you mean by `Git rev [filename].sch`. Only directories and Git revisions can be compared together.

I'm working on the `--relative` path option, and for this option, filesystem will show the relative path, in case it is provided on the command line.

leoheck, 2021-05-20:
> I don't understand what you mean by Git rev [filename].sch. Only directories and Git revisions can be compared together.

I see, check the new "commit revision" label in the second image in the first message on this theread. There you will find this text `filesystem .` Is this what you wanted? If yes, then this is not an issue at all.  If you think it is not a good message, then I am suggesting something different as "Git rev unstaged" indicating that image refers to the local version of the file, instead of something from the git history.

> I'm working on the --relative path option, and for this option, filesystem will show the relative path, in case it is provided on the command line.

Good! Let me know if you want me to test something.

jnavila, 2021-05-21:
The `filesystem .` label is expected by design even if it is not overly useful.

It comes from a mostly unknown feature of plotgitsch, where you can compare arbitrary directories by using the form `dir:<absolute-path>` instead of a revision:

```sh
# plotgitsch -ifirefox dir:/path/to/first/version dir:/path/to/second/version
```
This form is useful when you still want to use the diffing algorithm, but with third party version control systems.

In such case, the labels are `filesystem /path/to/first/version` and `filesystem /path/to/second/version`

leoheck, 2021-05-21:
I see. I am closing this then. Pretty nice feature by the way.

closed 2021-05-21 by leoheck