jnavila/plotkicadsch #55: Transparent bg is add only to the diff image

leoheck, 2021-05-20:
Only the final image has this transparent bg.
[image]

> The default viewer shows this composition of images with the diff image in the center.

Using my own `img-diff` here to store temporary images does not have the new background, unfortunattely =/

```
➜ plotgitsch HEAD~5 HEAD~4 --colors=FF0000:00FF00:000000:FFFFFF80
use compare (ImageMagick) between bitmaps between Git rev HEAD~5 and Git rev HEAD~4
Exporting...
sch1-board.svg (41c834)
sch2-board.svg (4f9c66)
```

Could you propagate this to the source image?

leoheck, 2021-05-20:
Ah, a flag to keep the originals would be super good to have! I am using my own `git-imgdiff` just to save these files.

leoheck, 2021-05-20:
This flag one could keep the originals too.
```
-k, --keep
by default, the svg diff files are deleted after launching the
viewer; this option lets the files in place after viewing them.
```

jnavila, 2021-05-21:
This change involves more rework than what I expected.

leoheck, 2021-05-21:
My bad!

I am following your changes. It looks like you have to change a lot of files for every small feature. Is this something related to the OCaml itself or the features you are changing are not small as I was thinking.

jnavila, 2021-05-21:
The span of changes is mostly due to the chain spanning from commandline argument to low level kicadlib stuff. It seems a lot of volume but it's not a big deal.

This one is more complicated, because the color scheme option is tied to the diffing process when it's internal. What the feature requests is a change in how the schematics are drawn, which is different. In fact, these should be two separated options.

leoheck, 2022-02-17:
Since this is pretty hard to implement I am going to close this issue now. This can be addressed externally in the `git-imagediff` script, for instance by just replacing the color in the generated images, directly.

closed 2022-02-17 by leoheck