jnavila/plotkicadsch #56: Exception (Invalid_argument "index out of bounds")

leoheck, 2021-05-26:
@jnavila did you ever seen something like this?
What this exception means?

```
➜ plotgitsch -m -l ./newTest-cache.lib 9646e03 2060795        
use compare (ImageMagick) between bitmaps between Git rev 9646e03 and Git rev 2060795
Exception (Invalid_argument "index out of bounds")
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '#End Library
'hrowing away line 'DRAW
'hrowing away line 'ENDDRAW
'hrowing away line 'ENDDEF
'hrowing away line '#
'hrowing away line '#End Library

```
It comes from this project here https://github.com/jorlee99/newTest

jnavila, 2021-05-27:
The files are mixed up with different line endings. This used to be fixed, but cae0e2054e41b37867442a60f7cfea8b3c34b186 introduced a regression. Anyway, mixing line endings is always asking for trouble. Be very careful on how Git is set up on Windows.

Anyway, this needs a fix.

leoheck, 2021-05-27:
Oh, I see. This is the same thing it is happening with my kdiff script. Well, thanks for sharing this. I tried to filter git repo converting line endings to workaround this but it didn't do much. I have to give it a second try to make sure I was doing it right.

commit e7f05248 (2021-05-27): plotgitsch: trim lines for CR characters

Sometimes line endings on Windows are with CR LF instead of simply
LF.

fixes #56
closed 2021-05-27 by jnavila