jnavila/plotkicadsch #57: Minor details in the produced image

leoheck, 2021-06-19:
Hey @jnavila do you have an idea why these small missmatches happen?

Some symbols are not pretty the same to what we see in the real schematic
 
Real symbol
[image]
Gitplotsch output
[image]

Other features are not equal too 

Real schematic. I feel this one could be improved a little bit, but I have not clue.
[image]
Gitplotsch output
[image]

The whole page, for instance
[image]
Generated page
[image]

Real text displacement. Ah maybe this one is due to the font mismatch and the text anchor is 
[image]
Gitplotsch output
[image]

Don't get me wrong, pltogitsch is pretty good. I am just asking to understand why these items are slightly different.

jnavila, 2021-06-19:
Text displacements and arcs'wrong orientation are bugs, mostly known but difficult to fix because plotgitsch must mimic kicad while using svg primitives but I don't have a clear specification of how kicad draws arcs for instance. After some tries and failures, I found an almost good solution, and moved on. I'll give it another try during the implementation of the kicad_sch format.

For general layout, this really an internal algorithm, so it can be enhanced. I don't plan to implement the templating system.

leoheck, 2021-06-19:
Cool, thanks for the explanation. Since plotgitsch is comparing 2 versions generated with the same tool doesn't need to be a perfect image of Kicad's schematic.

closed 2021-06-19 by leoheck