jnavila/plotkicadsch #58: Internal errors happening on nested projects

leoheck, 2021-07-04:
Hi @jnavila, I am trying to compare local changes using the latest version of the repo on a nested project as a user to do.

If I am inside the nested project and call plotgitsh as it is, I receive this.
```
➜ plotgitsch 
plotgitsch: internal error, uncaught exception:
            (Sys_error
              "./library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library: Too many levels of symbolic links")
            Raised by primitive operation at file "plotkicadsch/src/trueFs.ml", line 40, characters 21-39
            Called from file "plotkicadsch/src/trueFs.ml", line 50, characters 6-19
            Called from file "plotkicadsch/src/kicadDiff.ml", line 141, characters 20-42
            Called from file "cmdliner_term.ml", line 25, characters 19-24
            Called from file "cmdliner.ml", line 117, characters 32-39
```

This was fixed in the past, but now I started to see this again.
Maybe this was introduced here https://github.com/jnavila/plotkicadsch/commit/69a2cd4931e3fb192135ef38bc70939026d96ac8

leoheck, 2021-07-04:
I am tweaking the git-imgdiff to show the command line it is being invoked.
It looks like plotgitsh is passing only the 1st file 
```
➜ plotgitsch -r led-panel -m -l led-panel/board-cache.lib HEAD~1 HEAD~2
internal diff and show with /home/lheck/Dropbox/Documents/kdiff//git-imgdiff between Git rev HEAD~1 and Git rev HEAD~2

Command line:
/home/lheck/Dropbox/Documents/kdiff//git-imgdiff /tmp/diff_boarde35db5.svg

Exporting...
/tmp/diff_boarde35db5.svg -> diff_boarde35db5.svg
 -> 
cp: cannot stat '': No such file or directory
Errored with code 1
```

leoheck, 2021-07-04:
Another small (related) issue.
See the command line, the file is `board.sch`, but the images are eating the first letter `oard` 
Also, is it possible to add a separator between board name and the random number at the end?
Something like this `/tmp/to_board_1870c3.svg`

```
assoc-board on  main [!?] 
➜ plotgitsch -m -l board-cache.lib HEAD~1 dir:./       
use compare (ImageMagick) between bitmaps between Git rev HEAD~1 and file system ./

Command line:
/home/lheck/Dropbox/Documents/kdiff/git-imgdiff /tmp/from_oardc1fde9.svg /tmp/to_oard1870c3.svg

Exporting...
/tmp/from_oardc1fde9.svg -> sch1-oard.svg
/tmp/to_oard1870c3.svg -> sch2-oard.svg

Command line:
/home/lheck/Dropbox/Documents/kdiff/git-imgdiff /tmp/from_boardc9869a.svg /tmp/to_board7519a0.svg

Exporting...
/tmp/from_boardc9869a.svg -> sch1-board.svg
/tmp/to_board7519a0.svg -> sch2-board.svg
```

jnavila, 2021-11-03:
Hello,

I'm back on this project... Do you still face this issue? Could you propose a minimal reproduction repo?

Thanks

leoheck, 2021-11-03:
Oh GOOD! Yes, I still face this. I can setup something later and test it again to make sure I am still having problems with this.

leoheck, 2021-11-19:
Hi. I am experimenting a little bit more on this issue.
 
This repo is not nested. There is no extra .sch besides the schematics used inside the project. 

[image]

[image]

I think it is having problems trying to walk through symlinks that are not working. The issue that happened on this `./kiri/local/sch-HHKB_controller.svg` file that is a broken (red) symlink. And then, is related to the first message on this thread. The error there was related to symlinks too.

[image]

leoheck, 2021-11-19:
Yeah, this hypothesis seems right. 

In my tool, when I don't have an image, I am creating a symlink to a blank image. 

I need this because if one of the compared images is missing the comparison gets screwed in the HTML/javascript (I don't know how to handle this well, from the server-side, actually). Then I am creating a blank .svg image to fill those missing images when a schematic page or some layers are missing since they may not be present on all commits since the project changes. Also, to avoid duplications, I am creating symlinks instead. 

Experimenting here replacing symlinks with the blank images has solved the issue with `plotgitsch` where it fails on (maybe broken) symlinks.

jnavila, 2021-11-19:
Ah. Of course, files are expected to be real. Using dangling symlinks asks for trouble.

leoheck, 2021-11-19:
Considering symlinks may or may not exist and may or may not be good, is is possible to not fail when dealing with them?

leoheck, 2021-11-19:
Depending on how I set the -r parameter on this repo, the issues get weird
This repo has a local on root path library, it is shared with nested projects using symlinks.

[image]

```
led-board-repo/pir-board on  CU-4z3ga0-yb1.0-update-led-panel [?] 
➜ plotgitsch -r pir-board -k -m 5d48ef7
plotgitsch: internal error, uncaught exception:
            (Sys_error "./pir-board: No such file or directory")
            Raised by primitive operation at file "plotkicadsch/src/trueFs.ml", line 40, characters 21-39
            Called from file "plotkicadsch/src/trueFs.ml", line 50, characters 6-19
            Called from file "plotkicadsch/src/kicadDiff.ml", line 141, characters 20-42
            Called from file "cmdliner_term.ml", line 25, characters 19-24
            Called from file "cmdliner.ml", line 117, characters 32-39
```
```
led-board-repo/pir-board on  CU-4z3ga0-yb1.0-update-led-panel [?] 
➜ plotgitsch -r . -k -m 5d48ef7    
plotgitsch: internal error, uncaught exception:
            (Sys_error            
"././library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library/board/library: Too many levels of symbolic links")
            Raised by primitive operation at file "plotkicadsch/src/trueFs.ml", line 40, characters 21-39
            Called from file "plotkicadsch/src/trueFs.ml", line 50, characters 6-19
            Called from file "plotkicadsch/src/kicadDiff.ml", line 141, characters 20-42
            Called from file "cmdliner_term.ml", line 25, characters 19-24
            Called from file "cmdliner.ml", line 117, characters 32-39
```

jnavila, 2021-11-19:
Can you "ls -L 1 kicad-library" ?

jnavila, 2021-11-19:
The '-r' option allows to start the recursion at  a relative path to git root dir. In your case, it is useless. Using it with . (which is an absolute path) does not work.

leoheck, 2021-11-19:
Removing the symlinks, here it says it cannot find the project. this is how it looks 
[image]

```
led-board-repo
➜ tree -L 1
.
├── kicad-library
├── led-panel
└── README.md
```

> Can you "ls -L 1 kicad-library" ?

Sure but I don't have `-L`. I am assuming you use Mac, while I am using Linux.
There are no symlinks inside this kicad-library

leoheck, 2021-11-19:
I am running the script inside the project folder. 
Should I run it at the repo root level?

leoheck, 2021-11-19:
Make sense. Running it in the root level, wit the -r flag, works.

leoheck, 2021-11-19:
`.` is not an absolute path. it is relative, isn'r it? Absolute paths start from root `/` I assume.

jnavila, 2021-11-19:
Oh yes... `.` is a relative path. I don't think this is supported with `-r` because this is useless. If you're at `.` relative to the git root dir, well, you're at the git root dir and you don't need to specify.

Anyway, I'll try to add support for this "feature" :wink:

leoheck, 2021-11-19:
haha, nice.

So let me summarize it to see if I got it right.

The `-r` is relative to the repo root path.
But somehow it depends on where I am running the tool

So, for example, my repo is `my-repo` 2 kicad projects inside, let's call it proj1 and proj2
```
my_repo
├── proj1
└── proj2
```

So if I am in the root path `my_repo`. I could run it like
This seems to be working.
```
$ my_repo > 
plotgitsch -r proj1 commi1 commi2
```

I am inside one of the projects, *I cannot run it like this*
This is for sure an error because the path is wrong since it is relative to the repo's root path
```
$ my_repo/proj1 >
plotgitsch -r . commi1 commi2
```

Then I could use something like this. 
But now in this case I have an issue with the symlinks and it does not matter if they work or not.
If they work, it fails because I have somehow cyclic behavior (which is not the case in my projects)
If they don't work, it fails because something bad happens inside of plotgitsch
```
$ my_repo/proj1 > 
plotgitsch -r proj1 commi1 commi2
```

So, this last example is the thing that should be addressed inside plotgitsch, I think.

leoheck, 2021-11-25:
The same kind of problem happens when calling plotgitsch without parameters. It starts to look for something. Even if it is not a git repo. Or the main board folder/subfolder.

closed 2022-02-17 by leoheck