jnavila/plotkicadsch #59: Keeping original files even if they don't have changes

leoheck, 2021-07-04:
Sorry to bother you a lot. But something doesn't work anymore for me.

What I always needed is to keep original source .svg image files (for instance, file `from_` and file `to_`). For some reason, I am not able to save them anymore. I used to have a patch to save/export/generate them even if they didn't have any difference. The default behavior of the `pltogitsh` was always to exit if the schematics don't have changes. But I would like to keep these svgs even in this case. Having a command-line flag that allows this would be great, then I would not need to patch the original plotkicadsch. Unfortunately, it is so hard for me to make this proposal or to even implement it for myself. Then I made a patch instead.

The patch just removes this if statement here.
```
     | false ->
-      if textdiff then
         let cmd, args = diff_cmd F.label T.label @@ String.concat ~sep:"/" filename in
         SysAbst.exec cmd args >|= ignore
-      else Lwt.return ()
   in
```

Right now, maybe due to the latest changes, I cannot export these source .svg files anymore, they are not being generated. Maybe it is related with this issue actually, https://github.com/jnavila/plotkicadsch/issues/58

jnavila, 2021-11-03:
The if statement you remove is just the one that performs a text diff between files. It has nothing to do with keeping files.

leoheck, 2021-11-03:
Are you sure? Of course you are! But the goal was to generate svgs even if the files do not have any changes, and it looks like this fixed this issue for me. 

Could you check if it is possible to have a flag to "always" generate svg files?

jnavila, 2021-11-14:
Please be more precise about the behavior that you are expecting. The `-k option has rather different behaviors depending on the type of diff you ask for (external or internal). If you select the external diff utility, then pictures of both _old_ and _new_ versions of modified schematics are kept, but no _diff_ file is generated. If you use the internal diff, then the resulting diff file computed internally is kept.

leoheck, 2021-11-14:
I see. 

Well, for me, it does not matter which diff I am using. But I am using the internal diff, actually. I am also not using any diff image created by plotgitsh. I am using the generated source .svg for each side of the comparison, instead. Also, I am modifying the internal diff [git-imgdiff](https://github.com/leoheck/plotkicadsch/blob/master/plotkicadsch/src/git-imgdiff) (linked here is my version of it) since the external tool did not work as expected before. Something was not being generated.

I wanted to use your tool, as it is. And just changing the external tool. But for some reason, this did not work.  

With that being said, I would like to have the generated .svg files for both commits, in the same folder I am using the plotgitsch, and even if the images are the same. 

After, I am presenting the diff by using the same diff driver Kicad-Diff does. This is done entirely by CSS by overlapping 2 .svg images if I am not mistaken.

leoheck, 2021-11-14:
In summary, I am using plotgitsch to generate .svg images, for each schematic page in a Kicad project for both sides of the comparison. I am also discarding the final diff image.

jnavila, 2021-11-14:
In your case, this is the external diff utility (using git-imgdiff or a custom script) that you need to run. With the `-k` option, you will get the `from_*` and the `to_*` files for each version.

I ran a few tests with this configuration, and everything seems to run fine with current master branch. So, I guess you hit another corner case that I am unaware of.

leoheck, 2021-11-15:
It looks like everything is working good as you are describing except by this issue here.

Using this command on my script 
```
➜ plotgitsch  -k -m -l ./board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef   #&> /dev/null
```

I am sseing this 
```
plotgitsch: internal error, uncaught exception:
            (Sys_error "./kidiff/9e9d60c/board-02-In2_Cu.svg: No such file or directory")
            Raised by primitive operation at file "plotkicadsch/src/trueFs.ml", line 40, characters 21-39
            Called from file "plotkicadsch/src/trueFs.ml", line 50, characters 6-19
            Called from file "plotkicadsch/src/kicadDiff.ml", line 141, characters 20-42
            Called from file "cmdliner_term.ml", line 25, characters 19-24
            Called from file "cmdliner.ml", line 117, characters 32-39
```

Something is getting messed somewhere, but it is not my script, I suppose. Since plotgitsch is throwing the error.

When I run the command in the command line, it works without any issue.

```
assoc-board on  main [⇡!?] took 16s 
➜ plotgitsch  -k -m -l ./board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef   #&> /dev/null
use compare (ImageMagick) between bitmaps between Git rev fcd81ef and file system .

/home/lheck/Dropbox/Documents/kdiff/git-imgdiff from_board.svg to_board.svg

assoc-board on  main [⇡!?] 
➜ echo $?
0
```

Notice that the command line has only one side of the comparison since I want the comparison with the current working file.
Do you have any idea what this plotgisch output means?

leoheck, 2021-11-15:
Also, how do you specify the external diff utility? Right now I have one with the same name as `git-imgtool` and I am pre-pending to the PATH. But it is not good since I have to do it inside my script. I would like to use a different name, and put it in the path, and tell plotgitsch to use it instead. But there is no information about this on --help

leoheck, 2021-11-15:
By the external tool, you meant `git-imgtool` *only*, not any other tool name, right? Then in this case, how do I override the default `git-imgtool`? I have my version prepended in the PATH. 

```
➜ which git-imgdiff 
/home/lheck/Dropbox/Documents/kdiff//git-imgdiff
```

But when I run plotgitsch inside a script. It does not find my version of the external script.

jnavila, 2021-11-15:
You're right: `plotgisch` only runs `git-imgdiff`, which is the first one available in the `$PATH`. You can prepend PATH, with the path of the specific git-imgdiff that you created to make sure you run from another utility. I could add another option to specify the image diff utility that the user requests.

leoheck, 2021-11-15:
I am fine with preapending the path. But it looks it is not working as
expected since the sub shell (the script in this case) is not seeing this.
It is taking the git-imgdiff from opam folder. But it happens when it is
running inside a script. Let me check if I can so a simple script to
simulate this behavior.

jnavila, 2021-11-15:
I think you have found a bug. According to [the page about process management](https://ocsigen.org/lwt/4.1.0/api/Lwt_process#TYPEcommand):

> if the name is the empty string, then the first argument will be used. You should specify a name only if you do not want the executable to be searched in the PATH. On Windows the only way to enable automatic seach in PATH is to pass an empty name.

I specified the name, so the command is not searched in the `$PATH`. I'm going to push the fix to master. Can you check it?

leoheck, 2021-11-15:
Sure, I can test it. Can you let me know when you push this change?

leoheck, 2021-11-15:
It didn't work.

My environment

```
assoc-board on  main [⇡!?] took 45s 
➜ which plotgitsch
/home/lheck/.opam/4.09.1/bin/plotgitsch

assoc-board on  main [⇡!?] 
➜ which git-imgdiff 
/home/lheck/Dropbox/Documents/kdiff/git-imgdiff
```

When I run `plotgitsh` inside `kdiff` script
I am executing `which git-imgtool` before running `plotgitsh`

[image]

I am installing plotgitsch with your makefile
```
make build
make install
```

I just cannot see the version since the string is not quite right
```
assoc-board on  main [⇡!?] took 2m 9s 
➜ plotgitsch --version
%%VERSION%%
```

leoheck, 2021-11-15:
But this may kind of confirm I am using the last commit
[image]

leoheck, 2021-11-15:
Could you create an argument where I can pass the external tool as you mentioned before? I would like to test being able to explicitly tell the tool which external tool to use?

leoheck, 2021-11-15:
Ok, this issue is solved. It looks lie I was causing the issue by overriding the PATH again inside my script with this `eval "$(opam config env)"` on a specific situation.

closed 2021-11-15 by leoheck