jnavila/plotkicadsch #62: Diffing local changes does not work

leoheck, 2021-11-15:
Trying to make a diff with the local changes

```
assoc-board on  simplify-plotgitsch-usage [!?] 
➜ plotgitsch  -k -m -l .//board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef               
plotgitsch: internal error, uncaught exception:
            (Sys_error "./kidiff/9e9d60c/board-02-In2_Cu.svg: No such file or directory")
            Raised by primitive operation at file "plotkicadsch/src/trueFs.ml", line 40, characters 21-39
            Called from file "plotkicadsch/src/trueFs.ml", line 50, characters 6-19
            Called from file "plotkicadsch/src/kicadDiff.ml", line 141, characters 20-42
            Called from file "cmdliner_term.ml", line 25, characters 19-24
            Called from file "cmdliner.ml", line 117, characters 32-39
```

leoheck, 2021-11-15:
The error is this 125.

```
assoc-board on  simplify-plotgitsch-usage [!?] 
➜ plotgitsch  -k -m -l .//board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef   &> /dev/null

assoc-board on  simplify-plotgitsch-usage [!?] 
➜ echo $?
125
```

leoheck, 2021-11-15:
If I pass the board I am seeing this

```
➜ plotgitsch  -k -m -l ./board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef dir:board.sch                     

use compare (ImageMagick) between bitmaps between Git rev fcd81ef and file system board.sch
Exception (Invalid_argument "String.sub / Bytes.sub")
```

leoheck, 2021-11-17:
Experimenting on this again.

When I execute this, inside of a script with the `eval`. It does not work. 
```
eval plotgitsch -k -m -l ./board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef  &> /dev/null
echo $?
Status: 125
```

When I execute this on the command line it works
```
➜ plotgitsch -k -m -l ./board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef  &> /dev/null
echo $?
Status: 0
```

How do I debug this is `Called from file "plotkicadsch/src/kicadDiff.ml", line 141, characters 20-42`
There is a corner case that makes this fail to fail.
[image]

leoheck, 2021-11-17:
These are some tries to compare a version with the current local file 

```
assoc-board on  master [!?] 
➜ plotgitsch -k -m -l ./board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef                
plotgitsch: internal error, uncaught exception:
            (Sys_error "./kidiff/9e9d60c/board-02-In2_Cu.svg: No such file or directory")
            Raised by primitive operation at file "plotkicadsch/src/trueFs.ml", line 40, characters 21-39
            Called from file "plotkicadsch/src/trueFs.ml", line 50, characters 6-19
            Called from file "plotkicadsch/src/kicadDiff.ml", line 141, characters 20-42
            Called from file "cmdliner_term.ml", line 25, characters 19-24
            Called from file "cmdliner.ml", line 117, characters 32-39

assoc-board on  master [!?]
➜ plotgitsch -k -m -l ./board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef "board.sch"
use compare (ImageMagick) between bitmaps between Git rev fcd81ef and Git rev board.sch
Git Exception: cannot parse rev board.sch

assoc-board on  master [!?]
➜ plotgitsch -k -m -l ./board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef "dir:board.sch"
use compare (ImageMagick) between bitmaps between Git rev fcd81ef and file system board.sch
Exception (Invalid_argument "String.sub / Bytes.sub")

assoc-board on  master [!?]
➜ plotgitsch -k -m -l ./board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef .
use compare (ImageMagick) between bitmaps between Git rev fcd81ef and Git rev .
Git Exception: cannot parse rev .

assoc-board on  master [!?]
➜ plotgitsch -k -m -l ./board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef ..
use compare (ImageMagick) between bitmaps between Git rev fcd81ef and Git rev ..
Git Exception: cannot parse rev ..

assoc-board on  master [!?]
➜ plotgitsch -k -m -l ./board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef ...
use compare (ImageMagick) between bitmaps between Git rev fcd81ef and Git rev ../..
Git Exception: cannot parse rev ../..

assoc-board on  master [!?]
➜ plotgitsch -k -m -l ./board-cache.lib -c FF0000:008FFF:FFFFFF:222222 fcd81ef ....
use compare (ImageMagick) between bitmaps between Git rev fcd81ef and Git rev ../../..
```

leoheck, 2021-11-17:
Ah, I found something. When I have my generated `kidiff` folder inside the repo it fails when trying to find the local file. When I remove this folder. It manages to generate the source images of the diff process using only one side of the comparison and the local schematic file.   

I could simulate this ugly fix inside my script by moving the generated `kidiff` folder to another place outside of the directory of the repository, then right after executing `plotgitsch`, I am moving it back inside of the repo to do my stuff. It works flawlessly but it is not a good solution since I am doing this hundred times.  

So, any chance that plotgisch has a bug in the find local file method/function or anything related with having folders with many files or folders that are not being tracked?

leoheck, 2021-11-17:
A simple, not that good, fix would be to create a parameter to exclude some folders when walking through the repo, and then I would try to use it with my generated folder.

leoheck, 2021-11-17:
This workaround does not work when using nested projects. =/

jnavila, 2021-11-17:
The `dir:` prefix only accepts directory names, not schematic file names. This directory should match the same level as the current one in the git worktree. If you don't specify this prefix, then any option is interpreted as git revision specifier.

I introduced recursive search of `*.sch` files in the current directory, without taking into account that these files actually belong to the current project. This may have unwanted effects if you have schematic files unrelated to the current project.

leoheck, 2021-11-17:
Thanks for the explanation about this dir: tag. It was hard for me to understand it. Does the -help has this info?

Tried to find extra schematics, but there is none even inside my generated folders. It also looks like this problem is not happening anymore, at least in this repo that I have just one project on the root folder. It may be something that was misconfigured yesterday. I am closing this issue for now then.

closed 2021-11-17 by leoheck