jnavila/plotkicadsch #63: License type

leoheck, 2021-12-18:
Hi Jnavila, I was trying to add Kiri, which uses your tool on Kicad PCM. It probably won't work, as a submission since my tool has a lot of dependencies. But there is an issue that I don't understand much. It is about the licenses.

Plotgitsch has this license file. 
https://github.com/jnavila/plotkicadsch/blob/master/LICENSE.md

Which license is this one?

jnavila, 2021-12-19:
Hi,

If I remember correctly, it's a MIT-like license. I picked it up because I was also relying on some packages that had something similar and could not select a GPL one as such. I think it's no longer the case. Is it an issue for you?

leoheck, 2021-12-20:
Hi, thank you. 

This came up on this discussion here. Seth told me to fix the license field on my PR [here](https://gitlab.com/kicad/addons/metadata/-/merge_requests/27#note_784710740). And he also said I should use a license that fits both Plotgisch and Kicad-Diff. The latter has an MIT license but it was not clear the type of license in this repo. 

This one is Kicad-Diff's [LICENSE](https://github.com/Gasman2014/KiCad-Diff/blob/master/LICENSE) as reference.

closed 2021-12-31 by leoheck