jnavila/plotkicadsch #66: Plotkicadsh fails with missing components

leoheck, 2022-03-13:
Plotkicadsh fails with missing components.

The same behavior added to plotgitsch could be added to plotkicasch to ignore missing components.

```bash
➜  assoc-board git:(main) ✗ plotkicadsch -h           
plotkicadsch: unknown option '-h'.
plotkicadsch prints Kicad sch files to svg
  -l specify component library
  -f sch file to process
  -o full path of output directory
  -help  Display this list of options
  --help  Display this list of options

➜  assoc-board git:(main) ✗ plotkicadsch -f board.sch   
Fatal error: exception Kicadsch__Kicadlib.MakePainter(P).Component_Not_Found("power:+3V3")
```

jnavila, 2022-03-13:
For plotkicadsch, you must always specify the library.

leoheck, 2022-03-13:
Hm, then this will be a problem for me. 

I hacked eeschema sources to be able to convert the new .kicad_sch file to the legacy .sch. Now I would like to check what plot*sch can do for me to generate the svg. So, in this case, I don't have this lib.  

Here is where I share this fork of eeschema, if you are interested.
https://forum.kicad.info/t/solved-convert-the-new-kicad-sch-to-the-old-sch/34334/36?u=leoheck

jnavila, 2022-03-13:
If I understand, with the reverted format, you can only output the sch files but not the lib files. The problem with this is that the .sch files (unlike .kicad_sch files) do not contain the component descriptions.

leoheck, 2022-03-13:
I see. Then I have to investigate if it is possible to recreate this file
too.

closed 2022-04-01 by leoheck